chenxuuu/llcom #2: 希望能解析air202/air800的调试串口日志

wendal, 2019-06-27:
**您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**描述您想要的解决方案 / Describe the solution you'd like**
A clear and concise description of what you want to happen.

**描述您考虑过的替代方案 / Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**其他备注信息**
Add any other context or screenshots about the feature request here.

chenxuuu, 2019-06-27:
这个有啥意义？
实现能看日志，是不是还要加个下载烧录功能啊，有luatools的嘛

whc2001, 2019-06-27:
世界上有无数种串口设备（虽然作者看起来是按照合宙的API设计的），因此除非某个功能十分常用或请求人数很多，否则不可能为每一种都提供支持。
/* 个人觉得这种针对性比较强的功能做成插件比较合适，LLCOM增加多个Lua脚本后台异步执行以及允许Lua脚本通过API生成WPF窗体并增加控件，然后就可以用Lua写各种插件了。作者看起来做到目前这个状态就基本结束了，我个人对WPF不是很熟悉，如果有机会的话可能会看看怎么搞 */

closed 2019-06-28 by wendal