chenxuuu/llcom #6: win7 64bit系统上能打开串口，无法收发数据。

yuxiao, 2019-09-23:
**描述一下这个bug / Describe the bug**
win7 64bit系统上能打开串口，无法收发数据。

**复现步骤 / To Reproduce**
按照下面的步骤，可以复现bug / Steps to reproduce the behavior:
1. 打开串口正常（即使该串口已被其他程序占用，也能正常打开）
2. 无论是否打开串口，发送按钮都可以点击并在log窗口显示发送内容
3. 串口接收的内容无法在log窗口显示
4. 除最新版本外，历史版本也都存在这个问题

**预期的行为 / Expected behavior**
收发数据正常

**截图 / Screenshots**
无

**系统信息 / Desktop (please complete the following information):**
 - OS: Microsoft Windows 
 - Version: 6.1.7601

**其他备注信息 / Additional context**
Add any other context about the problem here.

chenxuuu, 2019-09-23:
请补充一下设备信息，有条件的话，录一个视频来体现上面所描述的问题

chenxuuu, 2019-09-23:
请补充一下设备信息，有条件的话，录一个视频来体现上面所描述的问题

 
  

串口接收的内容无法在log窗口显示

yuxiao, 2019-09-23:
没接设备，跟设备没关系。就是用两个USB转串口线，收发交叉连起来进行测试。

chenxuuu, 2019-09-23:
描述模糊，需要详细信息

yuxiao, 2019-09-24:
经过使用不同USB转串口线验证，该问题为USB转串口芯片的兼容性问题。
1. 使用FT4232H芯片的线，问题现象如以上描述，打开串口成功，无法收发数据（实际串口未打开）；
2. 使用FT232R芯片的线，无以上问题，收发正常；
3. 使用其他串口助手工具，包括LuaTools，用FT4232H芯片的线皆没有不兼容的问题。

chenxuuu, 2019-09-24:
已下单一块FT4232H板子，等到货之后测试

chenxuuu, 2019-09-26:
测试无问题
视频1：端口被占用后，状态正常，无法打开
视频2：串口数据互相通信无问题
[llcom1.mp4.zip](https://github.com/chenxuuu/llcom/files/3655888/llcom1.mp4.zip)
[llcom2.mp4.zip](https://github.com/chenxuuu/llcom/files/3655889/llcom2.mp4.zip)

chenxuuu, 2019-09-26:
[image]

closed 2019-09-27 by chenxuuu

yuxiao, 2019-09-30:
我把FT4232的线拿到win10 64bit系统的PC上试了，正如你的验证，工作正常。可以推断就是win7系统不兼容咯。

RYLF, 2020-04-10:
我也出现了你的问题。你最后就是换系统解决的吗？