chenxuuu/llcom #14: 长时间运行lua脚本后响应变慢

ttfaye, 2020-01-12:
**描述一下这个bug / Describe the bug**
打开一个llcom（v.1.0.2.6），运行lua脚本对收到的串口数据进行回送，刚开始时发出和回送间的延迟是120ms左右，llcom运行1个晚上后，回送时延变为300ms左右

**复现步骤 / To Reproduce**
按照下面的步骤，可以复现bug / Steps to reproduce the behavior:
1. 打开2个llcom程序，打开两个串口；
2. 两个串口rx、tx交叉loopback，用逻辑分析仪（示波器亦可）挂rx，tx，设置好信号采集及触发参数；
3. llcom程序a，运行lua脚本（见下文）回送收到的数据包；
4. llcom程序b，点发送，发送一段数据；
5. 从逻辑分析仪上观差收发见的时延；

**预期的行为 / Expected behavior**
持续运行较长时间lua脚本（如一个晚上）后，收发时延又120ms左右，延长至300ms；

**截图 / Screenshots**
如果需要，请放上你的截图。 / If applicable, add screenshots to help explain your problem.

**系统信息 / Desktop (please complete the following information):**
 - OS: win10_64
 - Version [1.0.2.6]

**其他备注信息 / Additional context**
```
运行的lua脚本：
--由于软件有自动保存
--所以请勿在开启本软件的同时
--用其他编辑器编辑此处打开了的脚本
--以免被覆盖掉
--建议在此处require你在改的脚本

--注册串口接收函数
uartReceive = function (data)
    log.info("uart receive",data)
    sys.publish("UART",data)--发布消息
end

--新建任务，等待接收到消息再继续运行
sys.taskInit(function()
    while true do
        --等待消息，超时1000ms
        local r,udata = sys.waitUntil("UART",1000)
        log.info("uart wait",r,udata)
        if r then
        	--local str=tostring(udata)
            --发送串口消息，并获取发送结果
            local sendResult = apiSendUartData("ok!"..tostring(udata))
            log.info("uart send",sendResult)
        end
    end
end)

--新建任务，每休眠1000ms继续一次
--sys.taskInit(function()
--    while true do
--        sys.wait(1000)--等待1000ms
--        log.info("task wait",os.time())
--    end
--end)

--1000ms循环定时器
--sys.timerLoopStart(log.info,1000,"timer test")
```

chenxuuu, 2020-01-12:
cpu和内存占用呢？

chenxuuu, 2020-01-12:
用快照版测试吧
https://ci.appveyor.com/project/chenxuuu/llcom/builds/30056006/artifacts

ttfaye, 2020-01-13:
cpu占用无异常，不过lua脚本运行较长时间，时延变长的时候，那个运行脚本的llcom，内存占用会跳动，例如从91.1M跳高到105.6M，又跳回91.xM；而刚开始运行lua脚本，时延较短的时候，llcom的内存占用是稳定不动的。

chenxuuu, 2020-01-13:
用快照版测过了吗？

whc2001, 2020-01-13:
感觉是垃圾回收的问题？之前做别的也遇到过类似的情况，不过这个因为涉及到Lua引擎可能难调一些

ttfaye, 2020-01-14:
多次测试，快照版在长时间运行lua后第一次发送，脚本应答回送的时延会较长，不过后续再发送，应答都正常了。可以认为快照版本解决了此问题。

closed 2020-01-14 by ttfaye