chenxuuu/llcom #17: 发送数据无反应

RYLF, 2020-04-09:
**描述一下这个bug / Describe the bug**
请使用简介并详细的语句，来描述这个bug。 / A clear and concise description of what the bug is.
打开串口后发送数据无反应

**复现步骤 / To Reproduce**
按照下面的步骤，可以复现bug / Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**预期的行为 / Expected behavior**
清楚简洁地描述你希望发生的事情。 / A clear and concise description of what you expected to happen.

**截图 / Screenshots**
如果需要，请放上你的截图。 / If applicable, add screenshots to help explain your problem.
[image]

**日志 / Logs**
有，请上传日志为附件 / Upload your log files.

**系统信息 / Desktop (please complete the following information):**
 - OS: [e.g. iOS]
win7 64
 - Version [e.g. 22]

**其他备注信息 / Additional context**
Add any other context about the problem here.

RYLF, 2020-04-09:
刚看了下，你这个软件好像是win7 64位不兼容哦。

chenxuuu, 2020-04-09:
每天都再win7上测的，怎么可能不行

RYLF, 2020-04-09:
> 每天都再win7上测的，怎么可能不行

那是什么原因呢。用SSCOM是可以的，用这个工具不行。

RYLF, 2020-04-10:
> 每天都再win7上测的，怎么可能不行

[image]

chenxuuu, 2020-04-10:
你能复现的话，能帮我测试吗？

RYLF, 2020-04-10:
> 你能复现的话，能帮我测试吗？

可以的，具体需要怎么测试。我的QQ：2461013399

chenxuuu, 2020-04-10:
@all-contributors please add @RYLF for bug

closed 2020-04-10 by chenxuuu

mikejzhiyu, 2020-04-16:
试试：发送的内容 加一下回车换行

commit fa65fa2d (2020-04-10): 更正奇怪的串口设备名导致打开错误串口号的问题 close #17