chenxuuu/llcom #19: 请增加“加回车换行”功能

mikejzhiyu, 2020-04-16:
**您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**描述您想要的解决方案 / Describe the solution you'd like**
A clear and concise description of what you want to happen.

**描述您考虑过的替代方案 / Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**其他备注信息**
Add any other context or screenshots about the feature request here.

chenxuuu, 2020-04-16:
工具自带了这个脚本的

更多设置 --> 对发送数据处理的Lua脚本 --> 脚本换成自带的那个**加上换行回车**

[image]

如果解决的话，点一下close

mikejzhiyu, 2020-04-16:
感谢回复。可以解决。
我用的是“快捷发送列表”的功能，误以为“加上换行回车”仅对Lua脚本有效。

closed 2020-04-16 by mikejzhiyu