chenxuuu/llcom #22: 针对IOT项目调试/测试的需求，建议添加以下需求

liusha0002000, 2020-12-04:
最近开发的项目中，涉及到串口和蓝牙BLE数据通讯，经常需要手动写代码来测试数据的收发，分别要在Windows和安卓上写代码来测试，但是效率和实用性比较低，这个工具用lua脚本来写，扩展开来的话，用作调试/测试工具，应该很方便，希望的扩展如下：

> 1、支持CRC计算，这个在串口传输时的数据校验经常用到；

> 2、支持二进制文件读写，在我的项目中会通过串口传声音数据，所以需要解析出来并保存成文件；

> 3、支持C库扩展，因为我的声音数据是opus编码的，这个编码提供的是C源码，如果此项目能支持通过lua调用第三方C库的API，那扩展性就很强了，这个需求可以另外开一个repo来专门说明吧；

> 4、支持BLE数据收发，在IOT项目也经常用到，Windows原生开发可以参考微软的例子：[https://github.com/microsoft/BluetoothLEExplorer](url)；

> 5、既然说到BLE支持，那就涉及到手机端开发了，考虑到跨平台开发的话，能选择的方案有微软的Xamarin: [https://docs.microsoft.com/zh-cn/xamarin/](url)和谷歌的Flutter: [https://flutterchina.club/docs/](url)，如果用Xamarin因为是C#开发，可能跟现有工程代码能有一定的复用，缺点是Xamarin在国内很少人用，用Flutter的话，就相当于开发一个全新的APP了，这个需要考虑好；

 以上，涉及蓝牙和跨平台开发可以一起来做，这个项目做成一个给嵌入式用的小型测试平台应该还是挺实用的；

chenxuuu, 2020-12-04:
1. 可以用lua写，不麻烦，没几行代码
2. 参考lua的io库
3. 理论上支持，需要测试
4. 不作支持，请使用微软的工具
5. 不考虑支持其他平台，手机没有跑串口调试工具的必要

chenxuuu, 2020-12-04:
刚刚试了，dll可以直接调用

[socket.zip](https://github.com/chenxuuu/llcom/files/5640476/socket.zip)

解压丢到llcom.exe同级目录

```lua
local r,info = pcall(function()
	socket=require("socket")
end)

if not r then
	print("load fail!",info:toHex())
else
	for i,j in pairs(socket) do
		print(i,j)
	end
end
```

liusha0002000, 2020-12-04:
OK
关于BLE，我先自己扩展试试看，其实这个跟串口通讯差不多，不过一般的工具都只是简单的数据收发，不能通过脚本来控制数据交互流程，我先自己改造下试试看；

closed 2020-12-04 by liusha0002000