chenxuuu/llcom #23: 希望添加发送文件的底层api

yuan-hp, 2020-12-21:
**您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
当数据量很大的时候，会将数据先放到文件，然后批量发送，这时候不希望等待很久才能发送完毕。
**描述您想要的解决方案 / Describe the solution you'd like**
A clear and concise description of what you want to happen.
在c#框架中直接开放出发送文件的api，让其在lua脚本中可调用。
**描述您考虑过的替代方案 / Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
我尝试直接使用lua读取文件发送，但是这样发送太慢，会消耗很长的时间。
**其他备注信息**
Add any other context or screenshots about the feature request here.

chenxuuu, 2020-12-21:
lua分段读再发也很慢吗，速度有多少，用的多大波特率？

chenxuuu, 2020-12-21:
刚才测试了一下发送1M文件数据，921600波特率

生成文件
```bash
dd bs=1M count=1 if=/dev/random of=test.bin
```

脚本
```lua
print(os.time())
local f = io.open([[D:\test.bin]],"rb")
local s
while 1 do
  s = f:read(1000)
  if not s then break end
  apiSendUartData(s)
end

f:close()
print(os.time())
print("done")
```

日志
```log
[10:52:25:9369][I]-[sys]	core loaded!
[10:52:25:9399]1608519145
[10:53:15:6435]1608519195
[10:53:15:6875]done
```

确实有些慢

在设置里把`显示发送的数据`关掉，就快了

```log
[10:55:45:2408][I]-[sys]	core loaded!
[10:55:45:2435]1608519345
[10:56:00:2007]1608519360
[10:56:00:2038]done
```

closed 2020-12-21 by yuan-hp