chenxuuu/llcom #29: llcom 会支持 Linux 系统吗？

taotieren, 2021-08-12:
# 如果不按该格式填写，开发者可能会直接忽略该issue

**您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
llcom 有计划支持 Linux 系统吗？
**描述您想要的解决方案 / Describe the solution you'd like**
A clear and concise description of what you want to happen.
vs 的工程可以用 linux 的 mono 的 xbuild 后可以在 linux 下运行，但是串口等功能会无法使用。这可能需要用[serialport-lib-dotnet](https://github.com/genielabs/serialport-lib-dotnet) 来兼容和转换
**描述您考虑过的替代方案 / Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**其他备注信息**
Add any other context or screenshots about the feature request here.

chenxuuu, 2021-08-12:
目前不考虑linux支持，如果你愿意做这些工作的话，欢迎pr

closed 2022-03-16 by chenxuuu

taotieren, 2022-10-27:
```bash
 makepkg -sf
==> 正在创建软件包：llcom-mono 1.1.1.1-0 (2022年10月27日 星期四 15时36分51秒)
==> 正在检查运行时依赖关系...
==> 正在检查编译时依赖关系==> 获取源代码...
  -> 找到 llcom-1.1.1.1.tar.gz
==> 正在验证 source 文件，使用sha256sums...
    llcom-1.1.1.1.tar.gz ... 通过==> 正在释放源码...
  -> 正在解压缩 llcom-1.1.1.1.tar.gz，使用 bsdtar
==> 正在删除现存的 $pkgdir/ 目录...
==> 正在开始 build()...
用于 Mono 的 Microsoft (R) 生成引擎版本 16.10.1
版权所有(C) Microsoft Corporation。保留所有权利。
在此解决方案中一次生成一个项目。若要启用并行生成，请添加“-m”开关。生成启动时间为 2022/10/27 15:36:52。节点 1 上的项目“/home/taotieren/git_clone/aur/llcom-mono/src/llcom-1.1.1.1/llcom.sln”(默认目标)。ValidateSolutionConfiguration:
  正在生成解决方案配置“Release|x64”。项目“/home/taotieren/git_clone/aur/llcom-mono/src/llcom-1.1.1.1/llcom.sln”(1)正在节点 1 上生成“/home/taotieren/git_clone/aur/llcom-mono/src/llcom-1.1.1.1/llcom/llcom.csproj”(2) (默认目标)。/home/taotieren/git_clone/aur/llcom-mono/src/llcom-1.1.1.1/llcom/llcom.csproj(440,5): error : 这台计算机上缺少此项目引用的 NuGet 程序包。使用“NuGet 程序包还原”可下载这些程序包。有关更多信息，请参见 http://go.microsoft.com/fwlink/?LinkID=322105。缺少的文件是 ../packages/NETStandard.Library.2.0.3/build/netstandard2.0/NETStandard.Library.targets。_CleanRecordFileWrites:
  正在创建目录“obj/x64/Release/”。已完成生成项目“/home/taotieren/git_clone/aur/llcom-mono/src/llcom-1.1.1.1/llcom/llcom.csproj”(默认目标)的操作 - 失败。项目“/home/taotieren/git_clone/aur/llcom-mono/src/llcom-1.1.1.1/llcom.sln”(1)正在节点 1 上生成“/home/taotieren/git_clone/aur/llcom-mono/src/llcom-1.1.1.1/WapProj/WapProj.wapproj”(3) (默认目标)。/home/taotieren/git_clone/aur/llcom-mono/src/llcom-1.1.1.1/WapProj/WapProj.wapproj(27,3): error MSB4019: 找不到导入的项目“/usr/lib/mono/xbuild/Microsoft/DesktopBridge/Microsoft.DesktopBridge.props”。请确认 Import 声明“/usr/lib/mono/xbuild/Microsoft/DesktopBridge//Microsoft.DesktopBridge.props”中的表达式正确，且文件位于磁盘上。已完成生成项目“/home/taotieren/git_clone/aur/llcom-mono/src/llcom-1.1.1.1/WapProj/WapProj.wapproj”(默认目标)的操作 - 失败。已完成生成项目“/home/taotieren/git_clone/aur/llcom-mono/src/llcom-1.1.1.1/llcom.sln”(默认目标)的操作 - 失败。
生成失败。
“/home/taotieren/git_clone/aur/llcom-mono/src/llcom-1.1.1.1/llcom.sln”(默认目标) (1) ->
“/home/taotieren/git_clone/aur/llcom-mono/src/llcom-1.1.1.1/llcom/llcom.csproj”(默认目标) (2) ->
(EnsureNuGetPackageBuildImports 目标) -> 
  /home/taotieren/git_clone/aur/llcom-mono/src/llcom-1.1.1.1/llcom/llcom.csproj(440,5): error : 这台计算机上缺少此项目引用的 NuGet 程序包。使用“NuGet 程序包还原”可下载这些程序包。有关更多信息，请参见 http://go.microsoft.com/fwlink/?LinkID=322105。缺少的文件是 ../packages/NETStandard.Library.2.0.3/build/netstandard2.0/NETStandard.Library.targets。

“/home/taotieren/git_clone/aur/llcom-mono/src/llcom-1.1.1.1/llcom.sln”(默认目标) (1) ->
“/home/taotieren/git_clone/aur/llcom-mono/src/llcom-1.1.1.1/WapProj/WapProj.wapproj”(默认目标) (3) ->
  /home/taotieren/git_clone/aur/llcom-mono/src/llcom-1.1.1.1/WapProj/WapProj.wapproj(27,3): error MSB4019: 找不到导入的项目“/usr/lib/mono/xbuild/Microsoft/DesktopBridge/Microsoft.DesktopBridge.props”。请确认 Import 声明“/usr/lib/mono/xbuild/Microsoft/DesktopBridge//Microsoft.DesktopBridge.props”中的表达式正确，且文件位于磁盘上。
    0 个警告    2 个错误
已用时间 00:00:00.62
==> 错误： 在 build() 中发生一个错误。    正在放弃...

```

通过 nuget 下载坑多，能改成静态编译吗？

taotieren, 2022-11-17:
https://github.com/actions/setup-dotnet

@chenxuuu  可以尝试下用 dotnet 环境来编译试试看。