chenxuuu/llcom #34: 有没有可能在lua代码中增加这些功能呢

wanglong126, 2021-11-30:
是否可以在lua代码区更改串口的设置呢
比如串口收到了某些消息,做好判断,通过某个方法修改串口的波特率,RTS,DTR等信息

chenxuuu, 2021-11-30:
可以直接读写公共变量，唯一问题就是可能会和UI上的显示不太同步

```lua
CS.llcom.Tools.Global.uart.Dtr = true
CS.llcom.Tools.Global.uart.Rts = true
print(CS.llcom.Tools.Global.uart.Rts)
print(CS.llcom.Tools.Global.uart.Dtr)

CS.llcom.Tools.Global.uart.Dtr = false
CS.llcom.Tools.Global.uart.Rts = false
print(CS.llcom.Tools.Global.uart.Rts)
print(CS.llcom.Tools.Global.uart.Dtr)

print(CS.llcom.Tools.Global.setting.baudRate)
print(CS.llcom.Tools.Global.setting.parity)
print(CS.llcom.Tools.Global.setting.dataBits)
print(CS.llcom.Tools.Global.setting.stopBit)
```

wanglong126, 2021-11-30:
CS.llcom.Tools.Global.uart.serial.BaudRate
CS.llcom.Tools.Global.setting.baudRate
修改这两个效果是一样的么?使用哪个更好一些呢

chenxuuu, 2021-11-30:
一个会同步更新配置文件和UI显示，另一个不会

wanglong126, 2021-11-30:
好的,感谢

closed 2021-11-30 by wanglong126