chenxuuu/llcom #35: 串口日志仅显示HEX

wanglong126, 2021-12-01:
串口目前有勾选是否显示HEX的选项
很多时候串口交互是纯HEX的,在通过代码进行解码
是否考虑给日志模式增加选项,仅显示HEX

比如日志如下
[2021-12-01 10:18:41.9613] ← ���
HEX:A1 1A A2 00 11 B1 1B
[2021-12-01 10:18:42.0032] → ���
HEX:A1 1A A2 00 11 B1 1B
[2021-12-01 10:18:42.9769] ← ���
HEX:A1 1A A2 00 11 B1 1B
[2021-12-01 10:18:43.0048] → ���
HEX:A1 1A A2 00 11 B1 1B
[2021-12-01 10:18:43.4253] ← ���
HEX:A1 1A A2 00 11 B1 1B
[2021-12-01 10:18:43.4472] → ���
HEX:A1 1A A2 00 11 B1 1B

SSCOM5显示如下
[10:21:17.968]发→◇A1 1A A2 00 11 22 B1 1B □
[10:21:17.970]收←◆A1 1A A2 00 11 22 B1 1B 
[10:21:18.304]发→◇A1 1A A2 00 11 22 B1 1B □
[10:21:18.307]收←◆A1 1A A2 00 11 22 B1 1B

chenxuuu, 2021-12-01:
试试快照版

wanglong126, 2021-12-01:
我下载了快照版,确实可以用这个实现我的需求,我的想法可以被实现
但是又有了几个新的`小问题`

首先附上原始的版本1.0.7.2收到的效果是这样的

[2021-12-01 13:45:56.7463] → ���P�
HEX:00 00 FF FF 01 00 D1 50 80

下载1.0.8.0快照版,如果不做任何修改,原始也是这样的

[2021-12-01 13:50:00.5342] → ���P�
HEX:00 00 FF FF 01 00 D1 50 80

-第一个问题
[image]
这里选择了接收的脚本,但是不会有效果
但是我如果在处理发送数据的脚本处更改了脚本选择项,会影响到接收的显示
如果处理发送选择的不是默认,日志会变成
[2021-12-01 13:53:00.0625] → 
HEX:

-第二个问题,也不算什么问题
如果串口对接收的数据日志展示修改为了hex,hex功能又存在,会显示成这样,在我看来这个不算bug存在,但是又觉得很怪异
[2021-12-01 13:53:48.0038] → 0000FFFF0100D15080
HEX:30 30 30 30 46 46 46 46 30 31 30 30 44 31 35 30 38 30
我理想中的结果是设置中对日志展示做下拉选择

-日志显示
--文本展示
--HEX展示
--文本和HEX展示

这样在程序中对`发送和接收`都做好内部处理

wanglong126, 2021-12-01:
特别想感谢一下作者,近期在使用中反馈的问题和想法基本都得到了实现,渐渐的基本可以仅使用llcom这一款串口工具完成我对串口的所有想法,可以在llcom中验证所有的串口交互,验证完成后再进行C#和合宙Luat的开发,感觉事半功倍,感谢!

chenxuuu, 2021-12-01:
试试快照版

wanglong126, 2021-12-01:
经过简单测试,已经很好的满足了我的需求
显示方式很满意,设置中的中文文本可以适当调整,接收的脚本测试保留功能后可以和发送的脚本这两个文本对称下
我这个问题可以结帖了,很满意

commit 3507e768 (2021-12-01): fix: #35 发送脚本匹配错误的问题
closed 2021-12-01 by wanglong126