chenxuuu/llcom #38: 增加注释快捷键

wanglong126, 2021-12-04:
代码区没有提示
可以按TAB和SHIFT TAB来调整格式
但是不能用SHIFT /来增加注释和取消注释
这个应该挺实用的

chenxuuu, 2021-12-04:
代码编辑器区域不打算再加新功能了，一般来说复杂代码都会用其他编辑器进行编辑，工具自带的编辑器只考虑基础高亮功能

wanglong126, 2021-12-06:
嗯,这样也是可以的,但现在使用外部编辑器和llcom的代码不同步,外部更改的llcom不会变,但是llcom更改的会覆盖掉外部的编辑器
我看到提示说尽可能代码用require引入,但实际使用的时候,我是将常用的lua脚本引入,实际开发的少量代码是在代码编辑区

最近在玩screeps,这个游戏的代码编辑区和vscode是同步的,两个地方无论修改哪一方,都会更新,我觉得这样很好

wanglong126, 2021-12-06:
git推送太痛苦了,附一下我的解决方法,

[image]

```c#
        Activated="Window_Activated"
        Deactivated="Window_Deactivated"
```

[image]

```c#
        //最后打开文件的时间
        private static DateTime lastLuaFileTime = DateTime.Now;
```

[image]

```c#
            //记录最后时间
            lastLuaFileTime = File.GetLastWriteTime(Tools.Global.ProfilePath + $"user_script_run/{Tools.Global.setting.runScript}.lua");
```
和
```c#
                //记录最后时间
                lastLuaFileTime = File.GetLastWriteTime(Tools.Global.ProfilePath + $"user_script_run/{fileName}.lua");
```

[image]

```C#
        private void Window_Deactivated(object sender, EventArgs e)
        {
            //窗口变为后台,可能在切换编辑器,自动保存脚本
            if (lastLuaFile != "")
                saveLuaFile(lastLuaFile);
        }
        private void Window_Activated(object sender, EventArgs e)
        {
            if (lastLuaFile != "")
            {
                //当前文件最后时间
                DateTime fileTime = File.GetLastWriteTime(Tools.Global.ProfilePath + $"user_script_run/{lastLuaFile}.lua");
                if (fileTime > lastLuaFileTime)//代码在外部被修改
                {
                    loadLuaFile(lastLuaFile);
                }
            }
        }
```

chenxuuu, 2021-12-06:
GitHub连不上的话，其实码云也有镜像，可以pr到那里
https://gitee.com/chenxuuu/llcom

chenxuuu, 2021-12-06:
@all-contributors please add @wanglong126 for bug, ideas and code

wanglong126, 2021-12-06:
> GitHub连不上的话，其实码云也有镜像，可以pr到那里 https://gitee.com/chenxuuu/llcom

不太熟练git类平台,在gitee上显示无权fok,就不太会如何操作了

chenxuuu, 2021-12-07:
@wanglong126 能fork了

wanglong126, 2021-12-07:
> @wanglong126 能fork了

试着提交了两个,也不知道对不对

closed 2022-03-01 by chenxuuu