chenxuuu/llcom #45: 首先感谢您 一款非常好的软件; 是否也可以给出一个 脚本调用 MQTT 的Demo  其实很重要呀

yudehe2009, 2022-04-12:
# 如果不按该格式填写，开发者可能会直接忽略该issue

**您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**描述您想要的解决方案 / Describe the solution you'd like**
A clear and concise description of what you want to happen.

**描述您考虑过的替代方案 / Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**其他备注信息**
Add any other context or screenshots about the feature request here.

yudehe2009, 2022-04-12:
非常好的一款软件  建议给出 脚本调用MQTT 协议的DEMO

chenxuuu, 2022-04-12:
目前没有给lua提供mqtt接口

yudehe2009, 2022-04-12:
感谢您百忙之中的回复 顺颂时祺！

closed 2022-07-29 by chenxuuu