chenxuuu/llcom #49: 运行lua脚本部分的一些优化建议

wanglong126, 2022-07-01:
# 如果不按该格式填写，开发者可能会直接忽略该issue

**您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

执行lua脚本后发现代码写的有一些问题,在外部编辑器已经做了修改
重新运行脚本,需要按下两次停止,然后再重新按下运行
流程感觉多了一些

**描述您想要的解决方案 / Describe the solution you'd like**
A clear and concise description of what you want to happen.

在lua脚本停止运行后,暂停按钮已经变得没有作用,改为重新运行感觉会更好

chenxuuu, 2022-07-01:
1.

https://github.com/chenxuuu/llcom/blob/master/llcom/DefaultFiles/user_script_run/example.lua#L1-L5

2. 停止运行后，给人留看输出打印日志的时间
3. 重新运行功能可以加

wanglong126, 2022-07-01:
> 

嗯,停止按钮我觉得可以保持不变
暂停按钮已经变得没有作用,改为重新运行感觉会更好

chenxuuu, 2022-07-20:
已合并

closed 2022-07-20 by chenxuuu