chenxuuu/llcom #52: how MonitorComm work

Layty, 2022-07-14:
想了解下这个实现原理,因为这个好像是要绑定进程的,这个dll 的实现原理可以跨平台到linux吗?

chenxuuu, 2022-07-14:
就是注入进程，替换掉默认的文件读写接口，实现接管串口数据收发的功能

理论上可以用同样的原理监控linux软件

Layty, 2022-07-14:
那应该是win32 api ,用c/c++实现的吗 ,我看调用有Marshal的字眼,还是用net实现的 ,
我可以用 c++调用 放到qt里面吗,有什么次数限制吗

chenxuuu, 2022-07-14:
这个dll没有开源😂

Layty, 2022-07-14:
这个不是大佬你写的吗,直接调用dll.
C写的DLL应该可以直接调用把 ,C#写到估计调用不了了

Layty, 2022-07-14:
lib文件还有吗,我看github上只有一个dll

chenxuuu, 2022-07-14:
没有，只有dll

Layty, 2022-07-14:
我看dll有大llcom的签名,以为是大佬写的

Layty, 2022-07-14:
句柄可以返回  ,但是串口助手一点击发送数据就卡住了,数据发不出去,回调也没有打印
关闭hook后 串口助手可以发送数据出去恢复

```

#include <iostream>
#include<windows.h>  // 必须包含 windows.h

enum State
{
    Disconnect = 2,
    Receive,
    Send
};

typedef struct
{
    unsigned char ComPort;
    unsigned char CommState;
    int FileHandle;
    int DataSize;
    /*[MarshalAs(UnmanagedType.ByValArray, SizeConst = 8192)]*/
    unsigned char *Data;
}Udata;

typedef  long(CALLBACK* TOnData)(long lParam);
//启动串口监视，返回bool
typedef int(CALLBACK* TMonitorComm)(unsigned long Pid, unsigned longComIndex, TOnData lpCallFunc);
//关闭串口监视，返回bool
typedef int(CALLBACK* TUnMonitorComm)();

long CALLBACK MyCallbackDelegate(long d)
{

    printf("MyCallbackDelegate ##### !\n");
    return 1;

    //Udata* dd = (Udata*)d;
    //switch (dd->CommState)
    //{
    //case Send:
    //    printf("Send!\n");
    //    break;
    //case Receive:
    //    printf("Receive!\n");
    //    break;
    //case Disconnect:
    //    printf("Disconnect!\n");
    //    break;
    //default:
    //    break;
    //};
    //if (dd->DataSize)
    //{
    //    for (int i = 0; i < dd->DataSize; i++)
    //        printf("%02X ", dd->Data[i]);
    //}
    //printf("\r\n");
    //return 1;
}

int main()
{

    HINSTANCE dllDemo = LoadLibrary("serial_monitor.dll");

    TMonitorComm MonitorCommptr=NULL;
    TUnMonitorComm UnMonitorCommptr = NULL;
    if (dllDemo) {
        MonitorCommptr = (TMonitorComm)GetProcAddress(dllDemo, "MonitorComm");
        UnMonitorCommptr = (TUnMonitorComm)GetProcAddress(dllDemo, "UnMonitorComm");
    }
    else {
        printf("GetProcAddress FAILED!\n");
    }

    if (UnMonitorCommptr != NULL)
    {
        printf("UnMonitorCommptr GET SUCCESS!\n");
        UnMonitorCommptr();
        printf("UnMonitorCommptr RUN SUCCESS!\n");
        //MonitorCommptr(8172, 1, MyCallbackDelegate);
    }

    if (MonitorCommptr != NULL)
    {
        printf("MonitorCommptr GET SUCCESS!\n");
        bool ret=MonitorCommptr(7952, 1, MyCallbackDelegate);
        //bool ret = MonitorCommptr(7952, 1, NULL);
        std::cout << "ret="<<ret<<"\n";
        printf("MonitorCommptr RUN  %d\n", ret);
        //MonitorCommptr(8172, 1, MyCallbackDelegate);
    }

    
    while (1);
    std::cout << "Hello World!\n";
}

```

Layty, 2022-07-15:
注册成功是-1吗  我更改参数有0或者-1

Layty, 2022-07-15:
这个有100次限制吗

chenxuuu, 2022-07-15:
为什么你会觉得有限制

Layty, 2022-07-15:
我用dumpbin 看到了 pcomm ,这个库后续版本有100限制,不知道是不是同一个底层库,
不知道为啥vs2019/c++/c  调用总是不成功

chenxuuu, 2022-07-15:
参考llcom源码调用

closed 2022-07-20 by chenxuuu