chenxuuu/llcom #64: 关闭软件再次打开后界面大小以及位置问题

linhongz, 2022-07-29:
**描述一下这个bug / Describe the bug**
调整软件界面大小以及在显示区域的位置后，关闭软件再次开启时，会先在默认位置以默认大小的方式显示出来，再闪现到自定义大小及自定义位置。

**复现步骤 / To Reproduce**
关闭软件重新打开，百分百复现

**预期的行为 / Expected behavior**
重新打开软件不会先在默认位置以默认大小显示，直接按照用户上次关闭软件前的位置和大小显示

**截图 / Screenshots**

https://user-images.githubusercontent.com/49241612/181668625-3c96600c-995d-4192-8ddc-a4cc392da02b.mp4

**日志 / Logs**
无

**系统信息 / Desktop (please complete the following information):**
 - OS: Windows10 x64
 - Version 1.1.0.6

chenxuuu, 2022-07-29:
两个解决方案，看你要哪个：

1. 不会跳，但是点开exe后需要灯两秒才出现窗口
2. 点开exe后立刻出现窗口，再恢复到上一次窗口大小与位置

linhongz, 2022-07-29:
第二个方案就是目前版本的样子吧

chenxuuu, 2022-07-29:
1. 例子exe [llcom.zip](https://github.com/chenxuuu/llcom/files/9215535/llcom.zip)
2. 就是现在的

可以尝试对比一下启动速度，看能接受哪一种

至于为什么选了2，因为有人建议用2

linhongz, 2022-07-29:
延迟时间很短，不会有明显的感觉，要我选的话就1好点

chenxuuu, 2022-07-29:
改为折衷方案

chenxuuu, 2022-07-29:
@all-contributors please add @linhongz for ideas and bug

commit 82a2d043 (2022-07-29): update: 读取配置文件的函数单独拎出来提前加载，fix #64
closed 2022-07-29 by chenxuuu