chenxuuu/llcom #69: 关于TCP功能

Layty, 2022-08-26:
# 如果不按该格式填写，开发者可能会直接忽略该issue

**您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**描述您想要的解决方案 / Describe the solution you'd like**
A clear and concise description of what you want to happen.

**描述您考虑过的替代方案 / Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**其他备注信息**
Add any other context or screenshots about the feature request here.

chenxuuu, 2022-09-29:
先设计一套接口的样式吧

目前串口的接收接口是`apiSendUartData`函数与`uartReceive`回调的形式，需要改一下

比如改成这样，channel可以用`mqtt`、`uart`、`tcp`等代替

```lua
apiSend(channel, data)
apiSetCb(channel, funcrion)
```

例子:

```lua
apiSetCb("mqtt", funcrion(d) print("recv from mqtt",d) end)
sys.timerLoopStart(apiSend,1000,"mqtt","test")
```

有什么建议吗

closed 2023-01-29 by chenxuuu

Layty, 2023-01-29:
建议api文档增加下socket相关的

chenxuuu, 2023-01-29:
着什么急，还没开工呢

commit f65a269a (2023-01-29): add: 通用消息通道接口的框架, close #69