chenxuuu/llcom #71: 希望可以在发送文本数据时支持转义字符

linhongz, 2022-09-05:
**您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.**
某些场景需要通过判断接收到了特定的转义字符来触发功能（例如 \r ）

**描述您想要的解决方案 / Describe the solution you'd like**
在发送文本数据时支持转义字符，SSCOM是支持的，如图（图中\x后面两个问号应该是显示乱码）

[image]

**描述您考虑过的替代方案 / Describe alternatives you've considered**
无

**其他备注信息**
Add any other context or screenshots about the feature request here.

chenxuuu, 2022-09-05:
使用自定义lua脚本可自行添加该类数据处理

chenxuuu, 2022-09-06:
https://github.com/chenxuuu/llcom/blob/master/llcom/DefaultFiles/user_script_send_convert/%E8%A7%A3%E6%9E%90%E6%8D%A2%E8%A1%8C%E5%9B%9E%E8%BD%A6%E7%9A%84%E8%BD%AC%E4%B9%89%E5%AD%97%E7%AC%A6.lua

已有例子，可自行修改添加功能

closed 2022-09-06 by chenxuuu