chenxuuu/llcom #81: Win7 运行工具，无法打开

Eminlin, 2022-11-08:
## 描述一下这个bug / Describe the bug
[image]

窗口提示的问题详细信息： 

问题签名:
  问题事件名称:	APPCRASH
  应用程序名:	llcom.exe
  应用程序版本:	1.1.1.2
  应用程序时间戳:	d372a854
  故障模块名称:	KERNELBASE.dll
  故障模块版本:	6.1.7600.16385
  故障模块时间戳:	4a5bdfe0
  异常代码:	e0434352
  异常偏移:	000000000000aa7d
  OS 版本:	6.1.7600.2.0.0.256.1
  区域设置 ID:	2052
  其他信息 1:	7b5a
  其他信息 2:	7b5a5a18cc5e9cee13c2ac15c385ef9f
  其他信息 3:	61c5
  其他信息 4:	61c53d9c389016393968d5e4f3740e7a

联机阅读隐私声明:
  http://go.microsoft.com/fwlink/?linkid=104288&clcid=0x0804

如果无法获取联机隐私声明，请脱机阅读我们的隐私声明:
  C:\Windows\system32\zh-CN\erofflps.txt

## 复现步骤 / To Reproduce
按照下面的步骤，可以复现bug / Steps to reproduce the behavior:
1. 下载软件运行，提示缺失 .net v4.0.3019
2. 前往 https://www.microsoft.com/zh-cn/download/confirmation.aspx?id=17718 下载
3. 

## 预期的行为 / Expected behavior
清楚简洁地描述你希望发生的事情。 / A clear and concise description of what you expected to happen.

## 截图 / Screenshots
如果需要，请放上你的截图。 / If applicable, add screenshots to help explain your problem.

## 日志 / Logs
无

## 系统信息 / Desktop (please complete the following information):

运行于虚拟机环境：

[image]

## 其他备注信息 / Additional context
暂无

chenxuuu, 2022-11-08:
依赖.net framework 4.6.2

closed 2022-11-08 by chenxuuu

Eminlin, 2022-11-08:
回复神速 感谢解答