chenxuuu/llcom #82: 发送Hex数据时，实际传输的数据与想要发送的数据不一致

jiajun2121, 2022-11-11:
## 描述一下这个bug / Describe the bug
请使用简介并详细的语句，来描述这个bug。 / A clear and concise description of what the bug is.

## 复现步骤 / To Reproduce
按照下面的步骤，可以复现bug / Steps to reproduce the behavior:
1. 发送框填入"B5 62"
2. 点击发送按钮

## 预期的行为 / Expected behavior
"处理发送数据的脚本"选择“16进制数据”时：应该发送2个字节"B5 62"(hex)，但是实际发送2个字节"B5 42"
"处理发送数据的脚本"选择“默认”时：应该发送5个字节"B5 62"(ascii)，但是实际发送2个字节"µb"

## 截图 / Screenshots
[image]
[image]

## 日志 / Logs
```
2022-11-11 16:49:28.948 +08:00 [INF] [SRART]Logs by LLCOM. https://github.com/chenxuuu/llcom
2022-11-11 16:49:28.985 +08:00 [DBG] [OpenClosePortButton]now:False
2022-11-11 16:49:28.986 +08:00 [DBG] [openPort]False,通信端口 (COM1)
2022-11-11 16:49:28.986 +08:00 [DBG] [openPort]GetPortNames
2022-11-11 16:49:28.986 +08:00 [DBG] [openPort]GetPortNames6
2022-11-11 16:49:28.986 +08:00 [DBG] [openPort]PortName:COM1,isOpeningPort:False
2022-11-11 16:49:28.988 +08:00 [DBG] [openPort]SetName
2022-11-11 16:49:28.988 +08:00 [DBG] [openPort]open
2022-11-11 16:49:28.988 +08:00 [DBG] [UartOpen]refreshSerialDevice
2022-11-11 16:49:28.991 +08:00 [DBG] [refreshSerialDevice]start
2022-11-11 16:49:28.991 +08:00 [DBG] [refreshSerialDevice]lastPortBaseStream.Dispose
2022-11-11 16:49:28.991 +08:00 [DBG] [refreshSerialDevice]BaseStream.Dispose
2022-11-11 16:49:28.992 +08:00 [DBG] [refreshSerialDevice]Dispose
2022-11-11 16:49:28.992 +08:00 [DBG] [refreshSerialDevice]new
2022-11-11 16:49:28.992 +08:00 [DBG] [refreshSerialDevice]done
2022-11-11 16:49:28.992 +08:00 [DBG] [UartOpen]open
2022-11-11 16:49:28.994 +08:00 [DBG] [UartOpen]done
2022-11-11 16:49:28.994 +08:00 [DBG] [openPort]change show
2022-11-11 16:49:28.995 +08:00 [DBG] [openPort]check to send
2022-11-11 16:49:28.995 +08:00 [DBG] [openPort]done
2022-11-11 16:49:28.995 +08:00 [DBG] [openPort]all done
2022-11-11 16:49:38.939 +08:00 [INF] <-µB
2022-11-11 16:49:38.939 +08:00 [DBG] [HEX]B5 42
2022-11-11 16:49:41.554 +08:00 [INF] <-µB
2022-11-11 16:49:41.554 +08:00 [DBG] [HEX]B5 42
2022-11-11 16:49:46.620 +08:00 [INF] <-µb
2022-11-11 16:49:46.620 +08:00 [DBG] [HEX]B5 62
```

## 系统信息 / Desktop (please complete the following information):
```
版本	Windows 10 专业版
版本号	21H1
安装日期	‎2021-‎6-‎16
操作系统内部版本	19043.1165
体验	Windows Feature Experience Pack 120.2212.3530.0
```

## 其他备注信息 / Additional context
LLCOM版本 ：1.1.1.5 及 1.1.1.6 （是的，V1.1.1.5发现bug之后重启软件正好提示更新V1.1.1.6，但是问题依旧(⓿_⓿)）

chenxuuu, 2022-11-11:
[image]

检查一下这里是否已经勾选

jiajun2121, 2022-11-11:
检查了一下，确实是勾选的原因，谢谢。

closed 2022-11-11 by chenxuuu