chenxuuu/llcom #85: 处理发送数据的脚本

medlarking, 2022-11-27:
## 描述一下这个bug / Describe the bug
处理发送数据的脚本，使用发送16进制数据return uartData:fromHex()，不管输入是大写还是小写，都按大写16进制发送，

## 复现步骤 / To Reproduce
按照下面的步骤，可以复现bug / Steps to reproduce the behavior:
1. Go to '主界面，更多设置，处理发送数据的脚本
2. 将默认处理语句改为：return uartData:fromHex()
3. 点击测试脚本，
4. 输入test 和TEST ，不能区分大小写，均按大写发送

## 预期的行为 / Expected behavior
区分大小写，按16进制正确发送代码

## 截图 / Screenshots
如果需要，请放上你的截图。 / If applicable, add screenshots to help explain your problem.

## 日志 / Logs
`更多设置`--`打开串口收发日志目录`，可以看到日志 / Upload your log files.

## 系统信息 / Desktop (please complete the following information):
 - OS: [e.g. iOS]
 - Version [e.g. 22]
 - 如果可以，可以附加配置文件内容（`更多设置`--`打开串口收发日志目录`--切换到上一级目录--`settings.json`文件）

## 其他备注信息 / Additional context
Add any other context about the problem here.

chenxuuu, 2022-11-27:
我没有看懂这个issue的意思，fromhex不就是要把数据转成hex吗，和大小写有什么关系？

另外test里的t和s完全就不是16进制数的部分

 

 
 
 
  

点击测试脚本，

medlarking, 2022-11-27:
字串test和TEST是不一样的字符，转成16进制数值应该是不一样的，

medlar
***@***.***

chenxuuu, 2022-11-27:
https://github.com/chenxuuu/llcom/blob/master/LuaApi.md#stringtohexstr-separator

https://github.com/chenxuuu/llcom/blob/master/LuaApi.md#stringfromhexhex

要不你再仔细看看接口文档？

chenxuuu, 2022-11-27:
不管是`TEST`还是`test`，都不是合法的16进制字符串

medlarking, 2022-11-27:
我想实现的功能是：字符串按16进制发送，源文本是test和TEST：

软件脚本处理完，全都成了54455354，肯定是哪里出了问题，要不费心再查查？

medlar
***@***.***

 
https://github.com/chenxuuu/llcom/blob/master/LuaApi.md#stringtohexstr-separator
 
https://github.com/chenxuuu/llcom/blob/master/LuaApi.md#stringfromhexhex

chenxuuu, 2022-11-27:
好吧，原来你连文档都懒得看，甚至我发出来你也不看

closed 2022-11-27 by chenxuuu