chenxuuu/llcom #95: Socket客户端发送换行

itldg, 2022-12-28:
## 描述一下这个bug / Describe the bug
目前软件发送换行,会发出`0D 0A`
是否可以通过选项,仅发出`0A`呢

## 复现步骤 / To Reproduce
按照下面的步骤，可以复现bug / Steps to reproduce the behavior:
如题

## 预期的行为 / Expected behavior
清楚简洁地描述你希望发生的事情。 / A clear and concise description of what you expected to happen.

## 截图 / Screenshots
如果需要，请放上你的截图。 / If applicable, add screenshots to help explain your problem.

## 日志 / Logs
`更多设置`--`打开串口收发日志目录`，可以看到日志 / Upload your log files.

## 系统信息 / Desktop (please complete the following information):
 - OS: [e.g. iOS]
 - Version [e.g. 22]
 - 如果可以，可以附加配置文件内容（`更多设置`--`打开串口收发日志目录`--切换到上一级目录--`settings.json`文件）

## 其他备注信息 / Additional context
Add any other context about the problem here.

chenxuuu, 2022-12-28:
socket客户端发送数据时，没有设置末尾数据的功能

itldg, 2022-12-28:
嗯,手动按回车会发出0D 0A,没有办法仅发0A,协议要求\n分割,有什么办法可以处理么

chenxuuu, 2022-12-28:
标准回车换行就是\r\n，实在要搞的话就用hex

itldg, 2022-12-28:
要发送的数据是这样的,HEX有点麻烦
[image]

提个建议在这里吧,如果有空了可以完善,之前用的这个是`\n`的
[image]

chenxuuu, 2022-12-28:
标准的换行就是\r\n

closed 2022-12-28 by chenxuuu