chenxuuu/llcom #99: 希望MQTT支持证书链接方式

jinhe1314, 2023-01-30:
## 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
希望MQTT支持证书链接方式
## 描述您想要的解决方案 / Describe the solution you'd like
A clear and concise description of what you want to happen.
类似MQTTX
## 描述您考虑过的替代方案 / Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.
无
## 其他备注信息
Add any other context or screenshots about the feature request here.
[image]

chenxuuu, 2023-02-01:
自选证书？没测试服务器，可以给个能测试的服务器和连接配置吗

chenxuuu, 2023-02-06:
测试服务器：
https://www.emqx.com/zh/mqtt/public-mqtt5-broker

测试用的客户端证书：
[certificate.zip](https://github.com/chenxuuu/llcom/files/10661838/certificate.zip)

commit 6fb0b7e8 (2023-02-06): add: mqtt tls手动指定证书功能, close #99
closed 2023-02-06 by chenxuuu