chenxuuu/llcom #100: 新版本的通用消息通道接口很棒,能否区分设置显示格式

itldg, 2023-02-01:
## 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.
比如 串口使用HEX日志,Socket客户端使用文本

## 描述您想要的解决方案 / Describe the solution you'd like
目前的通用消息接口,可以灵活的处理来自各个方面的数据
如果可以在设置中为不同的通道设置不同的日志格式就更棒了

另外比如Socket是否可以拿到链接成功和断开连接的消息呢

## 描述您考虑过的替代方案 / Describe alternatives you've considered
暂无

## 其他备注信息
暂无

chenxuuu, 2023-02-01:
区分日志格式，没想到什么好的处理办法，给用户设置起来也感觉很麻烦

连接和断开这种消息要加上也可以，反正传入的是table

itldg, 2023-02-01:
这个新版本体验很棒,已经可以用llcom模拟DTU的运行了,可以更加便利的开发代码了,省去了下载的时间,点赞支持

chenxuuu, 2023-02-01:
模拟DTU可还行，来贡献个代码⚡：[提交社区脚本](https://github.com/chenxuuu/llcom/discussions/87)

itldg, 2023-02-01:
体验很棒,但是这个不好分享,没什么共用性
我们是从设备上采集一些协议,然后解析,转成json后发给我们的平台
平台有监控之类的,通知DTU做下操作

等后续有什么通用类的,可以帮到大家的,我会分享脚本出来