chenxuuu/llcom #108: 串口数据接收过程中报错

itldg, 2023-02-10:
## 描述一下这个bug / Describe the bug
接收数据过程中频繁弹出错误
```
receive convert lua script error

System.Exception: XLua.LuaException: cannot open E:\Tools\llcom\uart_script_recv_convert/default.lua: No such file or directory
stack traceback:
	[C]: in function 'dofile'
	E:\Tools\llcom\core_script\once.lua:7: in function 'core_script.once'
	(...tail calls...)

   在 XLua.LuaEnv.ThrowExceptionFromError(Int32 oldTop)

   在 XLua.LuaEnv.DoString(Byte[] chunk, String chunkName, LuaTable env)

   在 XLua.LuaEnv.DoString(String chunk, String chunkName, LuaTable env)

   在 llcom.LuaEnv.LuaLoader.Run(String file, ArrayList args, String path)

   在 llcom.LuaEnv.LuaLoader.Run(String file, ArrayList args, String path)

   在 llcom.Pages.DataShowPage.DataUart..ctor(List`1 data, DateTime time, Boolean sent)
```

## 复现步骤 / To Reproduce
正常接收数据过程中,每10毫秒 5个字节

## 预期的行为 / Expected behavior
正常接收

## 截图 / Screenshots
[image]

## 日志 / Logs

截取了片段,剩下的都类似
```
2023-02-10 17:24:54.604 +08:00 [DBG] [HEX]0D 00 00 E0 03
2023-02-10 17:24:54.636 +08:00 [INF] ->
2023-02-10 17:24:54.636 +08:00 [DBG] [HEX]0E 00 00 10 03
2023-02-10 17:24:54.652 +08:00 [INF] ->A�
2023-02-10 17:24:54.652 +08:00 [DBG] [HEX]0F 00 00 41 C3
2023-02-10 17:24:54.684 +08:00 [INF] ->p
2023-02-10 17:24:54.684 +08:00 [DBG] [HEX]10 00 00 70 05
2023-02-10 17:24:54.701 +08:00 [INF] ->!�
2023-02-10 17:24:54.701 +08:00 [DBG] [HEX]11 00 00 21 C5
2023-02-10 17:24:54.733 +08:00 [INF] ->�c
2023-02-10 17:24:54.733 +08:00 [DBG] [HEX]00 B6 04 06 63
2023-02-10 17:24:54.749 +08:00 [INF] ->?�
2023-02-10 17:24:54.749 +08:00 [DBG] [HEX]3F 06 0B 01 C9
2023-02-10 17:24:54.780 +08:00 [INF] ->��
2023-02-10 17:24:54.780 +08:00 [DBG] [HEX]12 00 00 D1 C5
2023-02-10 17:24:54.796 +08:00 [INF] ->�
2023-02-10 17:24:54.796 +08:00 [DBG] [HEX]13 00 00 80 05
```

## 系统信息 / Desktop (please complete the following information):
win10

## 其他备注信息 / Additional context
另外是否可以给日志显示方式,增加一个`不记录日志`,串口数据都使用脚本功能处理了
数据很快,日志窗口一会儿就会满屏,然后又自动清理了,可以不需要显示的,这样处理数据应该更快吧?

chenxuuu, 2023-02-10:
`E:\Tools\llcom\uart_script_recv_convert/default.lua: No such file or directory`这是你缺文件了

不刷新日志模式已有，下面勾上就行

itldg, 2023-02-10:
> 

没有缺失的
[image]

程序运行过程中时不时的弹出

chenxuuu, 2023-02-10:
E:\Tools\llcom\uart_script_recv_convert

检查这里的文件

itldg, 2023-02-10:
[image]

chenxuuu, 2023-02-10:
那就很奇怪了，没看出问题，也不可能复现

itldg, 2023-02-10:
我复现了很多次,数据接收过程中就会时不时的弹出,弹出后串口界面就卡住不动了

需要的话明天我看看可以录个屏

closed 2023-02-11 by chenxuuu

chenxuuu, 2023-02-11:
原因不明，只能优化一下逻辑了，不排除是系统文件io接口的bug

commit a53777a5 (2023-02-11): fix: 单次运行脚本不再每次运行都读文件 close #108