chenxuuu/llcom #109: Socket 创建失败

itldg, 2023-02-11:
## 描述一下这个bug / Describe the bug
```
[2023/02/11 16:11:56.119]🚀 public socket server: 📢 Server is creating...
[2023/02/11 16:11:56.262]🚀 public socket server: 📢 Create failed
```

## 复现步骤 / To Reproduce
创建TCP和其他的

## 预期的行为 / Expected behavior
正常创建

## 截图 / Screenshots
[image]

## 日志 / Logs
无

## 系统信息 / Desktop (please complete the following information):
无

## 其他备注信息 / Additional context
无

itldg, 2023-02-11:
如果将代码中的
```csharp
       public WebSocket ws = new WebSocket("wss://netlab.luatos.com/ws/netlab");
       public WebSocket wsV6 = new WebSocket("wss://netlab.luatos.org/ws/netlab");
```
调整为
```csharp
      public WebSocket ws = new WebSocket("ws://netlab.luatos.com/ws/netlab");
      public WebSocket wsV6 = new WebSocket("ws://netlab.luatos.org/ws/netlab");
```
可以暂时解决问题

chenxuuu, 2023-02-11:
经测试，https://netlab.luatos.org的可以正常连接，另一个不能，原因不明

另外这功能和工具本身没有什么关联，只是用到了这两个网站而已

itldg, 2023-02-11:
> 经测试，https://netlab.luatos.org 的可以正常连接，另一个不能，原因不明
> 
> 另外这功能和工具本身没有什么关联，只是用到了这两个网站而已

我测试https://netlab.luatos.org 网页创建正常,llcom创建错误,断点测试

在发送创建消息时,socket连接未成功

发送消息会触发error,错误提示`An error has occurred in sending data.`

chenxuuu, 2023-02-11:
[image]

fixed

commit 6b3cf023 (2023-02-11): fix: #109
closed 2023-02-11 by chenxuuu