chenxuuu/llcom #111: 日志顺序不太理想

itldg, 2023-02-23:
## 描述一下这个bug / Describe the bug
收到数据后脚本答复,日志框会先显示答复,后显示收到的

## 复现步骤 / To Reproduce
收到数据,脚本答复

## 预期的行为 / Expected behavior
正确的顺序

## 截图 / Screenshots
[image]

## 日志 / Logs
不需要

## 系统信息 / Desktop (please complete the following information):
无

## 其他备注信息 / Additional context

另外日志箭头的方向是不是反了?

chenxuuu, 2023-02-23:
1. 红发绿收，无问题
[image]

2. 特性如此，sscom的箭头方向是反逻辑的。接收到的数据应指向数据本身，发送的数据应反方向指数据

closed 2023-02-23 by chenxuuu

itldg, 2023-02-23:
日志顺序反了呢
我使用sscom发送11给llcom,llcom收到数据后回复6F 6B 21
但是在llcom的日志上,先显示得到6F 6B 21,后显示的11

chenxuuu, 2023-02-23:
你需要证明这一点，比如同时间的逻辑分析仪波形数据

chenxuuu, 2023-02-23:
测试信息不完整，无法理解