chenxuuu/llcom #121: receive convert lua script error

itldg, 2023-03-13:
## 描述一下这个bug / Describe the bug
请使用简介并详细的语句，来描述这个bug。 / A clear and concise description of what the bug is.

[image]

弹窗原文

```
receive convert lua script error

System.Exception: XLua.LuaException: c# exception:System.IO.DirectoryNotFoundException: δ���ҵ�·����E:\Tools\llcom\tiggerCB\llcom\user_script_recv_convert\default.lua����һ���֡�

   �� System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)

   �� System.IO.FileStream.Init(String path, FileMode mode, FileAccess access, Int32 rights, Boolean useRights, FileShare share, Int32 bufferSize, FileOptions options, SECURITY_ATTRIBUTES secAttrs, String msgPath, Boolean bFromProxy, Boolean useLongPath, Boolean checkHost)

   �� System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options, String msgPath, Boolean bFromProxy, Boolean useLongPath, Boolean checkHost)

   �� System.IO.StreamReader..ctor(String path, Encoding encoding, Boolean detectEncodingFromByteOrderMarks, Int32 bufferSize, Boolean checkHost)

   �� System.IO.File.InternalReadAllText(String path, Encoding encoding, Boolean checkHost)

   �� FileWrap1.ReadAllText29(IntPtr L)
stack traceback:
	[C]: in field 'ReadAllText'
	[string "chunk"]:9: in function '!once!'

   在 XLua.LuaEnv.ThrowExceptionFromError(Int32 oldTop)

   在 XLua.LuaFunction.Call(Object[] args, Type[] returnTypes)

   在 llcom.LuaEnv.LuaLoader.Run(String file, ArrayList args, String path)

   在 llcom.LuaEnv.LuaLoader.Run(String file, ArrayList args, String path)

   在 llcom.Pages.DataShowPage.DataUart..ctor(List`1 data, DateTime time, Boolean sent)

```

## 复现步骤 / To Reproduce
按照下面的步骤，可以复现bug / Steps to reproduce the behavior:
串口每`100ms`毫秒接收一条串口消息

`运行脚本`功能使用注册串口接收,运行后就会报错

示例代码如下
```
uartReceive = function(data)
	log.info('hex',data:toHex())  
end
```

## 预期的行为 / Expected behavior
清楚简洁地描述你希望发生的事情。 / A clear and concise description of what you expected to happen.
没有弹窗

## 截图 / Screenshots
如果需要，请放上你的截图。 / If applicable, add screenshots to help explain your problem.
复现动图
[image]

## 日志 / Logs
`更多设置`--`打开串口收发日志目录`，可以看到日志 / Upload your log files.

## 系统信息 / Desktop (please complete the following information):
 - OS: [e.g. iOS]
 - Version [e.g. 22]
 - 如果可以，可以附加配置文件内容（`更多设置`--`打开串口收发日志目录`--切换到上一级目录--`settings.json`文件）

## 其他备注信息 / Additional context
Add any other context about the problem here.
如果出现了弹窗,不点击确定,`运行脚本`功能区日志输出还是正常的,但是左侧日志区域日志就暂停了

chenxuuu, 2023-03-13:
打开设置，看你的接收脚本写了什么

itldg, 2023-03-13:
没有动过
[image]

chenxuuu, 2023-03-13:
把系统信息的配置文件填上

itldg, 2023-03-13:
```
{"quickSendList":[[{"id":1,"text":"03 56 20 53 54 41 0D","hex":true,"commit":"监控V_STA"},{"id":2,"text":"03 56 20 4D 44 45 0D","hex":true,"commit":"服务模式V MDE"},{"id":3,"text":"03 58 20 46 0D","hex":true,"commit":"错误代码X_F"}],[{"id":1,"text":"EB020AE2E3202020202020A9FA0000EB030F202720463035202D302E30306D2F730000EB041020204C69676874204C6F6164202020200000","hex":false,"commit":"绝对楼层05"},{"id":2,"text":"EB020AE2E3202020202020A9FA0000EB030F202720462D32202D302E30306D2F730000EB041020204C69676874204C6F6164202020200000","hex":false,"commit":"显示楼层-2"}],[{"id":1,"text":"AT","hex":false,"commit":"测试连接命令"},{"id":2,"text":"AT+VERSION","hex":false,"commit":"查询——程序版本号"},{"id":3,"text":"AT+HELP","hex":false,"commit":"查询帮助信息"},{"id":4,"text":"AT+NAME","hex":false,"commit":"查询/设置——名称"},{"id":5,"text":"AT+DEFAULT","hex":false,"commit":"恢复默认设置"},{"id":6,"text":"AT+RESET","hex":false,"commit":"软件复位/重启"},{"id":7,"text":"AT+PIN","hex":false,"commit":"查询/设置——配对码"},{"id":8,"text":"AT+BAUD","hex":false,"commit":"查询/设置——波特率"},{"id":9,"text":"AT+COD","hex":false,"commit":"查询/设置——设备类型"},{"id":10,"text":"AT+ROLE","hex":false,"commit":"查询/设置——模块 SPP 主从模式"},{"id":11,"text":"AT+IAC","hex":false,"commit":"查询/设置——查询访问码"},{"id":12,"text":"AT+RNAME","hex":false,"commit":"查询远端蓝牙设备名称"},{"id":13,"text":"AT+INQM","hex":false,"commit":"查询/设置——查询访问模式"},{"id":14,"text":"AT+CMODE","hex":false,"commit":"查询/设置——连接模式"},{"id":15,"text":"AT+BIND","hex":false,"commit":" 查询/设置——绑定蓝牙地址"},{"id":16,"text":"AT+CLEAR","hex":false,"commit":"清除记忆地址"},{"id":17,"text":"AT+UARTMODE","hex":false,"commit":" 查询/设置——串口通讯模式"},{"id":18,"text":"AT+LADDR","hex":false,"commit":"查询——本地蓝牙地址"},{"id":19,"text":"AT+STATE","hex":false,"commit":"查询——蓝牙模块工作状态"},{"id":20,"text":"AT+INQ","hex":false,"commit":"搜索远端蓝牙设备"},{"id":21,"text":"AT+AUTOINQ","hex":false,"commit":"查询/设置——是否自动搜索远端蓝牙设备"},{"id":22,"text":"AT+INQC","hex":false,"commit":"取消查询——远端蓝牙设备"},{"id":23,"text":"AT+AUTOCONN","hex":false,"commit":"查询/设置——是否自动连接远端蓝牙设备"},{"id":24,"text":"AT+CONNECT34,81,F4,93,8E,7B","hex":false,"commit":"连接远端蓝牙设备"},{"id":25,"text":"AT+IPSCAN","hex":false,"commit":"查询/设置——寻呼扫描、 查询扫描参数"},{"id":26,"text":"AT+SENM","hex":false,"commit":" 查询/设置——安全、 加密模式"},{"id":27,"text":"AT+LOWPOWER","hex":false,"commit":"查询/设置——低功耗模式"},{"id":28,"text":"AT+SNIFF","hex":false,"commit":"查询/设置——Sniff 节能方式"},{"id":29,"text":"AT+ENABLEIND","hex":false,"commit":"查询/设置——Indication 上行指令"},{"id":30,"text":"AT+LSP","hex":false,"commit":"查询——蓝牙配对列表"},{"id":31,"text":"AT+RESETPDL","hex":false,"commit":"清除全部蓝牙配对列表"},{"id":32,"text":"AT+REMOVEPDL0","hex":false,"commit":"清除指定蓝牙配对记录"},{"id":33,"text":"AT+SUPERVISION","hex":false,"commit":"查询/设置——断线监测时长"}],[{"id":1,"text":"AF BE CD AA 55 3A 4E DD 06 B5 F1 2E 4B 4A 31 24 BF AD 1F ED","hex":true,"commit":"验证返回1"},{"id":2,"text":"AF BE CD B3 C0 9B CF B0 65 3E AE B0 93 6D D8 A1 0F A3 48 3B","hex":true,"commit":"验证返回2"},{"id":3,"text":"AFBECD6666666666666666666666666666666666","hex":true,"commit":"欺骗答复"},{"id":4,"text":"AF BE CD 48 5A 3D C7 52 81 41 2D 51 D8 01 AF 7F 05 47 2E F3","hex":true,"commit":"发送"}],[{"id":1,"text":"4D","hex":true,"commit":"MODULE"},{"id":2,"text":"63","hex":true,"commit":"UP"},{"id":3,"text":"62","hex":true,"commit":"ON"},{"id":4,"text":"61","hex":true,"commit":"OFF"},{"id":5,"text":"3E","hex":true,"commit":"GOON"},{"id":6,"text":"3C","hex":true,"commit":"GOBACK"},{"id":7,"text":"46","hex":true,"commit":"FUNCTION"},{"id":8,"text":"48","hex":true,"commit":"F"},{"id":9,"text":"0D","hex":true,"commit":"ENTER"},{"id":10,"text":"59","hex":true,"commit":"E"},{"id":11,"text":"64","hex":true,"commit":"DOWN"},{"id":12,"text":"58","hex":true,"commit":"D"},{"id":13,"text":"7E","hex":true,"commit":"CLEAR"},{"id":14,"text":"52","hex":true,"commit":"C"},{"id":15,"text":"66","hex":true,"commit":"B"},{"id":16,"text":"65","hex":true,"commit":"A"},{"id":17,"text":"67","hex":true,"commit":"-"},{"id":18,"text":"39","hex":true,"commit":"9"},{"id":19,"text":"38","hex":true,"commit":"8"},{"id":20,"text":"37","hex":true,"commit":"7"},{"id":21,"text":"36","hex":true,"commit":"6"},{"id":22,"text":"35","hex":true,"commit":"5"},{"id":23,"text":"34","hex":true,"commit":"4"},{"id":24,"text":"33","hex":true,"commit":"3"},{"id":25,"text":"32","hex":true,"commit":"2"},{"id":26,"text":"31","hex":true,"commit":"1"},{"id":27,"text":"30","hex":true,"commit":"0"},{"id":28,"text":"4D 4D 31 31 31","hex":true,"commit":"进监控"},{"id":29,"text":"4D 31 59","hex":true,"commit":"快速进监控"}],[{"id":1,"text":"02 46 46 41 34 41 39 44 36 03","hex":true,"commit":"查询输入输出"}],[{"id":1,"text":"example string","hex":false,"commit":"右击更改此处文字"},{"id":2,"text":"lua可通过接口获取此处数据","hex":false,"commit":"发送"},{"id":3,"text":"aa 01 02 0d 0a","hex":true,"commit":"Hex数据也能发"},{"id":4,"text":"此处数据会被lua处理","hex":false,"commit":"发送"},{"id":5,"text":"右击序号可以更改这一行的位置","hex":false,"commit":"发送"},{"id":6,"text":"00","hex":true,"commit":"发送"}],[{"id":1,"text":"01C7027F00001704","hex":true,"commit":"数据1"},{"id":2,"text":"01C7028300007411","hex":true,"commit":"数据2"},{"id":3,"text":"01C702870000660D","hex":true,"commit":"数据3"},{"id":4,"text":"01C7028B00005029","hex":true,"commit":"数据4"},{"id":5,"text":"01C7028F00004235","hex":true,"commit":"数据5"},{"id":6,"text":"01C7029300005D01","hex":true,"commit":"数据6"},{"id":7,"text":"01C7029700004F1D","hex":true,"commit":"数据7"},{"id":8,"text":"01C7029B00007939","hex":true,"commit":"数据8"},{"id":9,"text":"01C7029F00006B25","hex":true,"commit":"数据9"},{"id":10,"text":"01C702A300002631","hex":true,"commit [... cut ...] blish/topic","mqttSubscribeTopic":"your/subcribe/topic","quickListName0":"通力电梯","quickListName1":"西继迅达","quickListName2":"蓝牙模块AT指令","quickListName3":"MC2_B","quickListName4":"奥的斯","quickListName5":"蒂森MC2_B","quickListName6":"默纳克实时监控","quickListName7":"楼层显示","quickListName8":"基本参数","quickListName9":"默纳克外呼","tcpClientServer":"43.142.176.65","tcpClientPort":8787,"tcpClientProtocolType":0,"tcpServerPort":2333,"udpServerPort":2333,"MaxPackShow":204800,"MaxPacksAutoClear":2000,"LagAutoClear":true}
```

chenxuuu, 2023-03-13:
基本原因就是文件不存在，打不开这个文件

itldg, 2023-03-13:
如果我不`运行脚本`,始终就是正常的,我一旦点击了运行脚本,会出现这个弹窗

chenxuuu, 2023-03-13:
接收到串口数据，就会触发接受处理脚本，接收处理脚本加载失败，错误原因是文件不存在

itldg, 2023-03-13:
看提示说错误的文件路径是:`E:\Tools\llcom\tiggerCB\llcom\user_script_recv_convert\default.lua`
我本机实际的路径是:`E:\Tools\llcom\user_script_recv_convert\default.lua`

如果直接运行源代码,没有看到这个错误,但是会引发其他错误
[image]
[image]

chenxuuu, 2023-03-13:
怪

chenxuuu, 2023-03-13:
报错点应该是在https://github.com/chenxuuu/llcom/blob/master/llcom/LuaEnv/LuaLoader.cs#L74 处的`ReadAllText`  
然后这里的`file`看报错信息显示的是`E:\Tools\llcom\tiggerCB\llcom\user_script_recv_convert\default.lua`，错误路径

`file`变量是在https://github.com/chenxuuu/llcom/blob/master/llcom/LuaEnv/LuaLoader.cs#L109 传入设置的，按理说如果传入的`Tools.Global.ProfilePath + path + file`是错误路径的话，https://github.com/chenxuuu/llcom/blob/master/llcom/LuaEnv/LuaLoader.cs#L95 这一行就应该立即退出，但实际正常跑下去了，所以这个变量传入也没问题

这就很奇怪了，肯定哪里把`file`变量给改掉了

chenxuuu, 2023-03-13:
我把代码里的`file`变量改个随机的名字吧

chenxuuu, 2023-03-13:
试试快照版

itldg, 2023-03-13:
好的,现在测试

itldg, 2023-03-13:
问题依旧,但是弹窗中的两个路径我这都是不存在的

`E:\Tools\llcom\uartools\llcom\user_script_recv_convert\default.lua`

`E:\Tools\llcom\tiggerCB\llcom\user_script_recv_convert\default.lua`

```
---------------------------
Notice
---------------------------
receive convert lua script error

System.Exception: XLua.LuaException: c# exception:System.IO.DirectoryNotFoundException: δ���ҵ�·����E:\Tools\llcom\tiggerCB\llcom\user_script_recv_convert\default.lua����һ���֡�

   �� System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)

   �� System.IO.FileStream.Init(String path, FileMode mode, FileAccess access, Int32 rights, Boolean useRights, FileShare share, Int32 bufferSize, FileOptions options, SECURITY_ATTRIBUTES secAttrs, String msgPath, Boolean bFromProxy, Boolean useLongPath, Boolean checkHost)

   �� System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options, String msgPath, Boolean bFromProxy, Boolean useLongPath, Boolean checkHost)

   �� System.IO.StreamReader..ctor(String path, Encoding encoding, Boolean detectEncodingFromByteOrderMarks, Int32 bufferSize, Boolean checkHost)

   �� System.IO.File.InternalReadAllText(String path, Encoding encoding, Boolean checkHost)

   �� FileWrap1.ReadAllText29(IntPtr L)
stack traceback:
	[C]: in field 'ReadAllText'
	[string "chunk"]:9: in function '!once!'

   在 XLua.LuaEnv.ThrowExceptionFromError(Int32 oldTop)

   在 XLua.LuaFunction.Call(Object[] args, Type[] returnTypes)

   在 llcom.LuaEnv.LuaLoader.Run(String file, ArrayList args, String path)

   在 llcom.LuaEnv.LuaLoader.Run(String file, ArrayList args, String path)

   在 llcom.Pages.DataShowPage.DataUart..ctor(List`1 data, DateTime time, Boolean sent)
---------------------------
确定   
---------------------------

```
---------------------------
Notice
---------------------------
receive convert lua script error

System.Exception: XLua.LuaException: c# exception:System.IO.DirectoryNotFoundException: δ���ҵ�·����E:\Tools\llcom\uartools\llcom\user_script_recv_convert\default.lua����һ���֡�

   �� System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)

   �� System.IO.FileStream.Init(String path, FileMode mode, FileAccess access, Int32 rights, Boolean useRights, FileShare share, Int32 bufferSize, FileOptions options, SECURITY_ATTRIBUTES secAttrs, String msgPath, Boolean bFromProxy, Boolean useLongPath, Boolean checkHost)

   �� System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options, String msgPath, Boolean bFromProxy, Boolean useLongPath, Boolean checkHost)

   �� System.IO.StreamReader..ctor(String path, Encoding encoding, Boolean detectEncodingFromByteOrderMarks, Int32 bufferSize, Boolean checkHost)

   �� System.IO.File.InternalReadAllText(String path, Encoding encoding, Boolean checkHost)

   �� FileWrap1.ReadAllText29(IntPtr L)
stack traceback:
	[C]: in field 'ReadAllText'
	[string "chunk"]:9: in function '!once!'

   在 XLua.LuaEnv.ThrowExceptionFromError(Int32 oldTop)

   在 XLua.LuaFunction.Call(Object[] args, Type[] returnTypes)

   在 llcom.LuaEnv.LuaLoader.Run(String file, ArrayList args, String path)

   在 llcom.LuaEnv.LuaLoader.Run(String file, ArrayList args, String path)

   在 llcom.Pages.DataShowPage.DataUart..ctor(List`1 data, DateTime time, Boolean sent)
---------------------------
确定   
---------------------------

```

```

chenxuuu, 2023-03-13:
整个软件文件夹压缩打包发我qq吧

anewyee, 2023-03-17:
也复现了这个问题，提示未引用到实例

itldg, 2023-03-23:
我认为这个问题是上个版本  `fix: 脚本回调日志顺序问题` 引起的,我将代码修改回去,弹窗就没有了

commit a09f97a3 (2023-03-23): fix: 发送接受脚本执行前多试几次 close #121
closed 2023-03-23 by chenxuuu