chenxuuu/llcom #125: socket客户端增加TLS选项

yuxiao, 2023-03-30:
## 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
无法测试与服务器的TLS连接

## 描述您想要的解决方案 / Describe the solution you'd like
A clear and concise description of what you want to happen.
小工具里面“socket客户端”协议选项里面增加TLS选项

## 描述您考虑过的替代方案 / Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.
无

## 其他备注信息
Add any other context or screenshots about the feature request here.
无

chenxuuu, 2023-03-30:
有测试服务器吗？  
需要支持自定义证书吗？  
单向验证还是双向验证？  
双向验证的话有可以用来测试的服务器吗？

yuxiao, 2023-03-30:
暂没有自定义证书的需求，不过支持更好（跟MQTT支持TLS选项一样的）
测试服务器：tcpbin.com 端口：4243
用法见https://tcpbin.com

yuxiao, 2023-03-30:
网上找到一个类似功能的工具 https://www.cnblogs.com/guogangj/p/5182129.html

chenxuuu, 2023-05-30:
[image]

done

commit 884a3a45 (2023-05-30): add: tcp客户端ssl功能， close #125
closed 2023-05-30 by chenxuuu