chenxuuu/llcom #127: 能否搞个深色模式？

Atomseek, 2023-04-08:
## 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
看了很多软件，特别是功能很多的那些上古软件，都感觉有丶丑。群主这个风格挺简约的，要是有个深色模式就好了

## 描述您想要的解决方案 / Describe the solution you'd like
A clear and concise description of what you want to happen.
可以根据系统跟随改变主题色

## 描述您考虑过的替代方案 / Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.
无

## 其他备注信息
Add any other context or screenshots about the feature request here.
无

chenxuuu, 2023-05-25:
重构后可能会添加该功能

chenxuuu, 2024-06-13:
等新版本上面再加吧，目前的版本太难改了

closed 2024-06-13 by chenxuuu