chenxuuu/llcom #129: 能否添加使窗口固定在最前面的功能

apan-pro, 2023-04-19:
## 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
能否添加使窗口固定在最前面的功能
## 描述您想要的解决方案 / Describe the solution you'd like
A clear and concise description of what you want to happen.
在标题栏最小化按钮旁边或其他地方添加一个按钮使窗口锁定在最前面，再点一次按钮取消锁定。
## 描述您考虑过的替代方案 / Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.
无
## 其他备注信息
Add any other context or screenshots about the feature request here.
无

chenxuuu, 2023-04-19:
设置里有窗口置顶开关

apan-pro, 2023-04-19:
好的，感谢回复

chenxuuu, 2023-04-19:
开着吧，置顶按钮确实应该放到窗口上

closed 2023-05-25 by chenxuuu