chenxuuu/llcom #141: 关于请求添加清空曲线绘图区的api接口

we-worker, 2023-07-25:
### 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

   在曲线绘制功能中，查阅api后没有发现清除绘图区域的api，这使得我在绘制例如FFT结果曲线时，每次只能手动清空日志来实现绘图区清空，能否添加一个清除绘图区的api接口。

-----------------
### 描述您想要的解决方案 / Describe the solution you'd like
A clear and concise description of what you want to happen.

能否添加一个清除绘图区域的api接口，例如apiClearCurve()

-------------------
### 描述您考虑过的替代方案 / Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

也可以尝试清空日志的api

---------------
### 其他备注信息
Add any other context or screenshots about the feature request here.

感谢制作组的付出，十分感谢

chenxuuu, 2023-07-25:
添加1000个点把数据推出画面即可  
实际`清空按钮`就是这么做的

we-worker, 2023-07-25:
谢谢，明白了。