chenxuuu/llcom #143: 通过LUA脚本或者其他方式，实现TCP服务端自动收发消息回应客户端

Rocoer7, 2023-08-31:
### 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.

使用环境场景：
通过IICOM建立TCP服务端，实现自动收发消息回应客户端。

### 描述您想要的解决方案 / Describe the solution you'd like

通过LUA脚本或者其他方式实现功能

### 描述您想要的详细使用步骤描述 / Describe the solution you'd like to use in what way

1.当TCP客户端连接服务端时，脚本自动发送指定的消息给到客户端。
2.当客户端主动向服务端发送消息时，服务端根据客户端发的消息自动回应对应的消息。

### 其他备注信息或截图 / Add any other context or screenshots about the feature request here

如果能做到的话，甚至自动收发串口消息。

chenxuuu, 2023-10-13:
所需功能的接口皆满足，可自行编写功能脚本

chenxuuu, 2023-11-09:
```lua
apiSetCb("netlab",function (data)
  log.info(
    "netlab received",
    data.client,
    data.data)
if data.client == "connected" then
	apiSend("netlab",nil,
	{
	  client = data.data,
	  data   = "hello"
	})
else
    --处理别的
	apiSend("netlab",nil,
	{
	  client = data.client,
	  data   = "hello"
	})
end
end)
```

commit 2ee5f66d (2023-11-09): add: lua接口增加 客户端连接/断开事件, close #143
closed 2023-11-09 by chenxuuu