chenxuuu/llcom #149: 执行带require"bit"模块的脚本时提示loading失败

gigss, 2023-10-30:
### 描述一下这个bug / Describe the bug

试过编译5.1 5.2和5.3的bit.dll并放置到llcom的根目录下,都会提示loading失败
[17:09:37:5070][I]-[sys]	core loaded!
[17:09:37:5110]lua代码报错了：
XLua.LuaException: error loading module 'bit' from file 'D:\Program Files\llcom\bit.dll':
	%1 ������Ч�� Win32 Ӧ�ó���

stack traceback:
	[C]: in ?
	[C]: in upvalue 'oldrequire'
	D:\Program Files\llcom\core_script\head.lua:41: in function 'require'
	D:\Program Files\llcom\user_script_run\test.lua:2: in main chunk
	[C]: in upvalue 'oldrequire'
	D:\Program Files\llcom\core_script\head.lua:41: in function 'require'
	[string "chunk"]:1: in main chunk
   在 XLua.LuaEnv.ThrowExceptionFromError(Int32 oldTop)
   在 XLua.LuaEnv.DoString(Byte[] chunk, String chunkName, LuaTable env)
   在 XLua.LuaEnv.DoString(String chunk, String chunkName, LuaTable env)
   在 llcom.LuaEnv.LuaRunEnv.<>c__DisplayClass16_0.<New>b__0()

### 复现步骤 / To Reproduce

单独执行lua命令,是可以导入bit库.
D:\Program Files\luabitop-53plus-master\luabitop-53plus-master>lua -v
Lua 5.1.5  Copyright (C) 1994-2012 Lua.org, PUC-Rio

D:\Program Files\luabitop-53plus-master\luabitop-53plus-master>lua bitbench.lua
loop baseline      9.2 ns
tobit             25.8 ns
bnot              25.1 ns

### 预期的行为 / Expected behavior

帮忙指导下怎样才能正确导入bit库

### 系统 / System

win10

### 软件版本 / Version

1.1.3.2

### 验证

- [X] 检查该问题是否已被提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.

chenxuuu, 2023-10-30:
把dll传上来

gigss, 2023-10-30:
> 把dll传上来

公司网络上传文件不方便，已发邮件给您，感谢回复🤝🏻

chenxuuu, 2023-10-30:
[image]

llcom的lua是64位的，你要用64位的dll才行

gigss, 2023-10-30:
> [image]
> 
> llcom的lua是64位的，你要用64位的dll才行
重新编译64位dill后提示这个
[20:02:20:7053][I]-[sys]	core loaded!
[20:02:20:7093]lua代码报错了：
XLua.LuaException: error loading module 'bit' from file 'D:\Program Files\llcom\bit.dll':
	�Ҳ���ָ����ģ�顣

stack traceback:
	[C]: in ?
	[C]: in upvalue 'oldrequire'
	D:\Program Files\llcom\core_script\head.lua:41: in function 'require'
	D:\Program Files\llcom\user_script_run\test.lua:2: in main chunk
	[C]: in upvalue 'oldrequire'
	D:\Program Files\llcom\core_script\head.lua:41: in function 'require'
	[string "chunk"]:1: in main chunk
   在 XLua.LuaEnv.ThrowExceptionFromError(Int32 oldTop)
   在 XLua.LuaEnv.DoString(Byte[] chunk, String chunkName, LuaTable env)
   在 XLua.LuaEnv.DoString(String chunk, String chunkName, LuaTable env)
   在 llcom.LuaEnv.LuaRunEnv.<>c__DisplayClass16_0.<New>b__0()

gigss, 2023-10-30:
我是通过附件源码编译bit.dll的

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2023年10月30日 18:09 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [chenxuuu/llcom] 执行带require"bit"模块的脚本时提示loading失败 (Issue #149) |

llcom的lua是64位的，你要用64位的dll才行

chenxuuu, 2023-10-30:
文件发出来

 
 
 
 | 发件人 | ***@***.***&gt; |
 | 日期 | 2023年10月30日 18:09 |
 | 收件人 | ***@***.***&gt; |
 | 抄送至 | ***@***.***&gt;***@***.***&gt; |

gigss, 2023-10-31:
> 文件发出来
> […](#)
> ---原始邮件--- 发件人: ***@***.***&gt; 发送时间: 2023年10月30日(周一) 晚上8:11 收件人: ***@***.***&gt;; 抄送: ***@***.******@***.***&gt;; 主题: Re: [chenxuuu/llcom] 执行带require"bit"模块的脚本时提示loading失败 (Issue #149) 我是通过附件源码编译bit.dll的

邮件转发您了,有空帮我看看哈

chenxuuu, 2023-10-31:
[image]

dll有问题

gigss, 2023-10-31:
> [image]
> 
> dll有问题

您那边有llcom能用的bit.dll吗,有的话麻烦发我试试

gigss, 2023-10-31:
> > [image]
> > dll有问题
> 
> 您那边有llcom能用的bit.dll吗,有的话麻烦发我试试
我这边用用5.3.4是可以导入的
[image]

chenxuuu, 2023-11-01:
我用不到bit库，因为lua5.3自带位运算功能

gigss, 2023-11-01:
> 我用不到bit库，因为lua5.3自带位运算功能
> […](#)

新版本可以把bit运行加入到新功能吗,串口数据解释经常要用到位运算呢

chenxuuu, 2023-11-01:
lua里直接就能用位运算符号，没有加的意义

gigss, 2023-11-01:
> lua里直接就能用位运算符号，没有加的意义
> […](#)

对哦,明白了,感谢