chenxuuu/llcom #153: About Language switching problem

mt6595, 2024-02-26:
### 描述一下这个bug / Describe the bug

打开/关闭串口和串口状态显示在切换语言的时候存在BUG,参考下图
[image]

### 复现步骤 / To Reproduce

1、打开或者关闭串口
2、修改语言

### 预期的行为 / Expected behavior

全英文显示或者全中文显示

### 系统 / System

Windows 10

### 软件版本 / Version

1.1.1.8

### 验证

- [X] 检查该问题是否已被提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.

LuaRobot, 2024-02-26:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction
- ❌未提供出错时需要的日志文件 does not provide logs when bug happening

chenxuuu, 2024-02-26:
重新打开软件即可

 

 
 LLCOM.png (view on web)

closed 2024-06-13 by chenxuuu