chenxuuu/llcom #154: Win10下编译报错 APPX3217 ，以及Tuple报错

asuper0, 2024-03-05:
### 描述一下这个bug / Describe the bug

1.Win10编译报错

`APPX3217 无法找到包含“UAP 10.0.22000.0”中的“UAP.props”的 SDK 文件夹`

折腾了很久，发现是WapProj这个项目的目标版本是Win11，我修改为Win10后可以编译了，建议默认修改为Win10。

2. [MainWindow.xaml.cs](https://github.com/chenxuuu/llcom/blob/master/llcom/View/MainWindow.xaml.cs#L1331) 和 [OnlineScriptsPage.xaml.cs](https://github.com/chenxuuu/llcom/blob/master/llcom/Pages/OnlineScriptsPage.xaml.cs#L140) 我这里编译会报错 

`error CS0121: 以下方法或属性之间的调用具有二义性:“System.TupleExtensions.Deconstruct<T1, T2>(System.Tuple<T1, T2>, out T1, out T2)”和“System.TupleExtensions.Deconstruct<T1, T2>(System.Tuple<T1, T2>, out T1, out T2)”`

修改成[这样的写法](https://github.com/chenxuuu/llcom/blob/master/llcom/View/MainWindow.xaml.cs#L1225) 就可以了，不知道你那边怎么可以编译，我的VS是2022且更新到最新了，框架是.Net Framework 4.8

### 复现步骤 / To Reproduce

克隆，编译

### 预期的行为 / Expected behavior

编译通过

### 系统 / System

Win10 22H2

### 软件版本 / Version

1.1.3.3

### 验证

- [X] 检查该问题是否已被提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.

LuaRobot, 2024-03-05:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供出错时需要的日志文件 does not provide logs when bug happening

chenxuuu, 2024-03-05:
WapProj无需编译，是微软商店的打包工程

新版（从五年前开始吧大概）.net框架和语法自带tuple支持，无需手动声明可以自动推断类型

ci编译完全无问题，所以查查你的环境吧

closed 2024-03-05 by chenxuuu