chenxuuu/llcom #159: 确认是显示界面吞HEX格式字符了

qq19538, 2024-03-13:
### 描述一下这个bug / Describe the bug

[image]
显示界面，把0x20 以及0x0A都吞掉了，猜测应该是ASCII中0x20表示空格，0x0A表示换行，HEX显示时，显示界面程序没有把字符做替换。

### 复现步骤 / To Reproduce

更改编码格式不能避免这个显示吞字符的问题。

### 预期的行为 / Expected behavior

[image]
实际LOG是这样的

### 系统 / System

WIN10

### 软件版本 / Version

1.1.3.3

### 验证

- [X] 检查该问题是否已被提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.

qq19538, 2024-03-13:
[log20240313.txt](https://github.com/chenxuuu/llcom/files/14586483/log20240313.txt)
日志文件如上

qq19538, 2024-03-13:
[image]
[image]
验证出问题在哪里，默认的脚本使用了fromHex()函数，这个函数应该会替换0x20以及0x0A这类的ASCII码。

chenxuuu, 2024-03-14:
软件默认脚本不会有任何处理：
https://github.com/chenxuuu/llcom/blob/master/llcom/DefaultFiles/user_script_recv_convert/default.lua
https://github.com/chenxuuu/llcom/blob/master/llcom/Model/Settings.cs#L29

fromhex不要乱用，这是用来把hex转成bin的函数

closed 2024-03-14 by chenxuuu