chenxuuu/llcom #160: 窗口滚动条莫名其妙会自动跳变至前面某一位置

hbl0307106015, 2024-03-23:
### 描述一下这个bug / Describe the bug

输出内容达到了数千行，此时往往需要拖动滚动条来查看之前的打印，但是出现一个现象：每隔一段时间，窗口滚动条莫名其妙会自动跳变至前面某一位置，导致我要反复拖动滚动条

### 复现步骤 / To Reproduce

输出内容达到了数千行

### 预期的行为 / Expected behavior

滚动条不会自动跳变

### 截图 / Screenshots

#暂无

### 日志 / Logs

#暂无

### 系统 / System

Win10

### 软件版本 / Version

1.1.3.3

### 验证

- [X] 检查该问题是否已被提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.

LuaRobot, 2024-03-23:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction
- ❌未提供出错时需要的日志文件 does not provide logs when bug happening

chenxuuu, 2024-05-03:
如何复现？

S4FR, 2024-05-24:
这个我也遇到过，就是输出超级多行，约一万行吧，然后关闭串口，拉到最下面，然后按住左键拖动鼠标，想要复制一些报文，然后就会弹上去了。比如我正在复制最底部的1%，可能会弹到20%的地方。

closed 2024-06-13 by chenxuuu

hbl0307106015, 2024-10-24:
@chenxuuu 这个问题解决了吗？