chenxuuu/llcom #161: 添加一个窗口置顶功能。

suimua, 2024-04-07:
### 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.

需要在不同软件界面切换，再去看llcom的数据时，要去任务栏把llcom界面调出来，有点不方便。

### 描述您想要的解决方案 / Describe the solution you'd like

给llcom添加一个窗口置顶选项

### 描述您想要的详细使用步骤描述 / Describe the solution you'd like to use in what way

在软件状态栏添加一个checkbox，用于决定是否要把llcom界面置顶。

LuaRobot, 2024-04-07:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供功能需求的详细描述 does not provide detailed description of the feature request

suimua, 2024-04-07:
突然发现已经有这个功能了，不好意思。

closed 2024-04-07 by suimua