chenxuuu/llcom #162: udp-server增加到lua脚本

www220, 2024-04-11:
### 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.

apiSetCb没有接受到udp的数据报文，导致udp的通讯测试无法写脚本，求大佬支持

### 描述您想要的解决方案 / Describe the solution you'd like

apiSetCb支持udp

### 描述您想要的详细使用步骤描述 / Describe the solution you'd like to use in what way

send支持udp

LuaRobot, 2024-04-11:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供功能需求的详细描述 does not provide detailed description of the feature request

chenxuuu, 2024-05-03:
udp的发和收是独立的两个页面，发的话有本地socket通道，收还没加，有空加上

onlyistranger, 2024-07-03:
同求支持udp的接收回调，方便udp测试

closed 2026-11-06 by chenxuuu