chenxuuu/llcom #164: 虚拟串口发送数据失败

SmartElec, 2024-05-14:
### 描述一下这个bug / Describe the bug

串口是使用vspd创建的虚拟串口,打开串口没有报错,发送数据的时候报错,报错信息如图

### 复现步骤 / To Reproduce

首先打开串口,发送框是默认的数据uart data
然后点击发送
弹出提示框
```
串口数据发送失败！请检查连接！
System.NullReferenceException：未将对象引I用设置到对象的实例。
在 llcom.Model.Uart.SendData(Bytel] data)
在llcom.MainWindow.sendUartData(Bytel] data,Nullable1is_hex)
```

### 预期的行为 / Expected behavior

预期的行为是不会出现发送失败

### 截图 / Screenshots

[image]

### 系统 / System

Win10 22H2

### 软件版本 / Version

1.1.3.3

### 验证

- [X] 检查该问题是否已被提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.

LuaRobot, 2024-05-14:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction
- ❌未提供出错时需要的日志文件 does not provide logs when bug happening

SmartElec, 2024-05-14:
上面说错了,是所有串口都是一样的效果.

chenxuuu, 2024-05-14:
信息太少了，提供一下下面的信息吧：

1. 软件报时是完整的截图
2. 整个文件夹打包发上来

SmartElec, 2024-05-14:
> 信息太少了，提供一下下面的信息吧：
> 
> 1. 软件报错时完整的截图
> 2. 整个文件夹打包发上来

[image]
[llcom.zip](https://github.com/chenxuuu/llcom/files/15303548/llcom.zip)

日志
```
2024-05-14 14:05:42.359 +08:00 [INF] [SRART]Logs by LLCOM. https://github.com/chenxuuu/llcom
2024-05-14 14:05:42.397 +08:00 [DBG] [OpenClosePortButton]now:False
2024-05-14 14:05:42.398 +08:00 [DBG] [openPort]False,USB Serial Port (COM18)
2024-05-14 14:05:42.398 +08:00 [DBG] [openPort]GetPortNames
2024-05-14 14:05:42.399 +08:00 [DBG] [openPort]GetPortNames4
2024-05-14 14:05:42.399 +08:00 [DBG] [openPort]PortName:COM18,isOpeningPort:False
2024-05-14 14:05:42.399 +08:00 [DBG] [openPort]SetName
2024-05-14 14:05:42.400 +08:00 [DBG] [openPort]open
2024-05-14 14:05:42.400 +08:00 [DBG] [UartOpen]refreshSerialDevice
2024-05-14 14:05:42.401 +08:00 [DBG] [refreshSerialDevice]start
2024-05-14 14:05:42.401 +08:00 [DBG] [refreshSerialDevice]lastPortBaseStream.Dispose
2024-05-14 14:05:42.401 +08:00 [DBG] [refreshSerialDevice]BaseStream.Dispose
2024-05-14 14:05:42.401 +08:00 [DBG] [refreshSerialDevice]Dispose
2024-05-14 14:05:42.401 +08:00 [DBG] [refreshSerialDevice]new
2024-05-14 14:05:42.401 +08:00 [DBG] [refreshSerialDevice]done
2024-05-14 14:05:42.401 +08:00 [DBG] [UartOpen]open
2024-05-14 14:05:42.405 +08:00 [DBG] [UartOpen]done
2024-05-14 14:05:42.405 +08:00 [DBG] [openPort]change show
2024-05-14 14:05:42.406 +08:00 [DBG] [openPort]check to send
2024-05-14 14:05:42.406 +08:00 [DBG] [openPort]done
2024-05-14 14:05:42.406 +08:00 [DBG] [openPort]all done
2024-05-14 14:07:46.963 +08:00 [DBG] [OpenClosePortButton]now:True
2024-05-14 14:07:46.963 +08:00 [DBG] [OpenClosePortButton]close
2024-05-14 14:07:46.963 +08:00 [DBG] [UartClose]refreshSerialDevice
2024-05-14 14:07:46.963 +08:00 [DBG] [refreshSerialDevice]start
2024-05-14 14:07:46.963 +08:00 [DBG] [refreshSerialDevice]lastPortBaseStream.Dispose
2024-05-14 14:07:46.963 +08:00 [DBG] [refreshSerialDevice]BaseStream.Dispose
2024-05-14 14:07:46.963 +08:00 [DBG] [refreshSerialDevice]Dispose
2024-05-14 14:07:46.963 +08:00 [DBG] [refreshSerialDevice]new
2024-05-14 14:07:46.963 +08:00 [DBG] [refreshSerialDevice]done
2024-05-14 14:07:46.963 +08:00 [DBG] [UartClose]Close
2024-05-14 14:07:46.963 +08:00 [DBG] [UartClose]done
2024-05-14 14:07:46.963 +08:00 [DBG] [OpenClosePortButton]close done
2024-05-14 14:07:46.963 +08:00 [DBG] [OpenClosePortButton]change show
2024-05-14 14:07:46.964 +08:00 [DBG] [OpenClosePortButton]change show done
2024-05-14 14:07:48.247 +08:00 [DBG] [OpenClosePortButton]now:False
2024-05-14 14:07:48.247 +08:00 [DBG] [openPort]False,USB Serial Port (COM18)
2024-05-14 14:07:48.247 +08:00 [DBG] [openPort]GetPortNames
2024-05-14 14:07:48.247 +08:00 [DBG] [openPort]GetPortNames4
2024-05-14 14:07:48.247 +08:00 [DBG] [openPort]PortName:COM18,isOpeningPort:False
2024-05-14 14:07:48.247 +08:00 [DBG] [openPort]SetName
2024-05-14 14:07:48.247 +08:00 [DBG] [openPort]open
2024-05-14 14:07:48.247 +08:00 [DBG] [UartOpen]refreshSerialDevice
2024-05-14 14:07:48.247 +08:00 [DBG] [refreshSerialDevice]start
2024-05-14 14:07:48.247 +08:00 [DBG] [refreshSerialDevice]lastPortBaseStream.Dispose
2024-05-14 14:07:48.247 +08:00 [DBG] [refreshSerialDevice]BaseStream.Dispose
2024-05-14 14:07:48.247 +08:00 [DBG] [refreshSerialDevice]Dispose
2024-05-14 14:07:48.247 +08:00 [DBG] [refreshSerialDevice]new
2024-05-14 14:07:48.248 +08:00 [DBG] [refreshSerialDevice]done
2024-05-14 14:07:48.248 +08:00 [DBG] [UartOpen]open
2024-05-14 14:07:48.251 +08:00 [DBG] [UartOpen]done
2024-05-14 14:07:48.251 +08:00 [DBG] [openPort]change show
2024-05-14 14:07:48.252 +08:00 [DBG] [openPort]check to send
2024-05-14 14:07:48.252 +08:00 [DBG] [openPort]done
2024-05-14 14:07:48.252 +08:00 [DBG] [openPort]all done

```

chenxuuu, 2024-05-14:
你的发送脚本怎么是空的，默认是`return uartData`

closed 2024-05-14 by chenxuuu