chenxuuu/llcom #171: 本地TCP服务器第一次打开报错,将无法再打开

itldg, 2024-07-03:
### 描述一下这个bug / Describe the bug

首次打开提示错误
```
通常每个套接字地址（协议/网络地址/端口）只允许使用一次。
```

关闭外部的端口占用后点击打开将无反应

### 复现步骤 / To Reproduce

占用端口,打开TCP服务端,取消占用端口,再打开TCP服务端

### 预期的行为 / Expected behavior

正常打开TCP服务端

### 系统 / System

Win10

### 软件版本 / Version

1.1.3.7

### 验证

- [X] 检查该问题是否已被提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.

LuaRobot, 2024-07-03:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction
- ❌未提供出错时需要的日志文件 does not provide logs when bug happening

itldg, 2024-07-05:
已在 #172 解决

closed 2024-07-05 by itldg