chenxuuu/llcom #174: 在win11下报病毒

li371359001, 2024-08-22:
### 描述一下这个bug / Describe the bug

我测试了几次，在win11下都会被系统查出毒而删掉

### 复现步骤 / To Reproduce

直接运行就会被删

### 预期的行为 / Expected behavior

系统下正常工作，不被删

### 系统 / System

win11 23h2

### 软件版本 / Version

1.1.3.7

### 验证

- [X] 检查该问题是否已被提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.

LuaRobot, 2024-08-22:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction
- ❌未提供出错时需要的日志文件 does not provide logs when bug happening

chenxuuu, 2024-08-22:
非软件自身问题

可使用微软商店版，已交保护费

closed 2024-08-22 by chenxuuu