chenxuuu/llcom #179: 串口数据窗口滑动不顺畅

Masterfans, 2024-09-28:
### 描述一下这个bug / Describe the bug

串口数据很多的时候用鼠标滑动视觉效果很卡，实际滑动应该是一页数据，当一页一行的时候就最后的数据可能因为滑动一页导致无法显示出来，只能通过鼠标操作滚动条才能滚动到指定的位置

### 复现步骤 / To Reproduce

串口接收到很多页数据，然后鼠标滑动，只能一页一页的滑动，如果是用的触摸板的话，很容易一下滚动了好几页的数据，无法做到只滚动几行，查看多页数据不方便

### 预期的行为 / Expected behavior

预期应该是文本的滚动随着鼠标滚动，对于多页数据有割裂的效果，观察数据不方便

### 截图 / Screenshots

第一页位置
[image]
鼠标触发一次滚动位置
[image]
鼠标拖动滚动条可以到达的位置
[image]

### 系统 / System

win11 23H2

### 软件版本 / Version

1.1.3.7

### 验证

- [X] 检查该问题是否已被提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.

LuaRobot, 2024-09-28:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction
- ❌未提供出错时需要的日志文件 does not provide logs when bug happening

chenxuuu, 2025-01-25:
已恢复老的显示逻辑

closed 2025-01-25 by chenxuuu