chenxuuu/llcom #181: 只显示文本，UTF-8编码，中文乱码的问题

Kuno1998, 2024-12-18:
### 描述一下这个bug / Describe the bug

当串口手动UTF-8编码的中文时，就会显示乱码

### 复现步骤 / To Reproduce

选择更多设置->串口数据显示方式->只显示文本数据。然后收到中文后就会乱码。如果改成同时显示文本和HEX中文就不会乱码

### 预期的行为 / Expected behavior

只显示文本数据时 不出现中文乱码。

### 系统 / System

W11

### 软件版本 / Version

1.1.3.7

### 验证

- [X] 检查该问题是否已被提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.

LuaRobot, 2024-12-18:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction
- ❌未提供出错时需要的日志文件 does not provide logs when bug happening

chenxuuu, 2024-12-18:
[image]

无问题

closed 2024-12-18 by chenxuuu