chenxuuu/llcom #183: https://github.com/chenxuuu/llcom/issues/new/choose

jinwuzhijing2020, 2024-12-22:
### 描述一下这个bug / Describe the bug

窗口保持置顶显示功能，需要每次点击才行，软件不记录状态。

### 复现步骤 / To Reproduce

窗口保持置顶显示功能，需要每次点击才行，软件不记录状态。

### 预期的行为 / Expected behavior

窗口保持置顶显示功能，需要每次点击才行，软件不记录状态。

### 截图 / Screenshots

[image]

### 系统 / System

Windows 11

### 软件版本 / Version

1.1.3.7

### 验证

- [X] 检查该问题是否已被提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.

LuaRobot, 2024-12-22:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction
- ❌未提供出错时需要的日志文件 does not provide logs when bug happening

chenxuuu, 2024-12-24:
小问题，有时间再修👀

commit 47cf5512 (2025-01-21): fix: 设置里窗口置顶重新开启不生效 fix #183
closed 2025-01-21 by chenxuuu