chenxuuu/llcom #184: 自定义波特率BUG

mt6595, 2024-12-29:
### 描述一下这个bug / Describe the bug

1、选择自定义波特率时，弹出波特率设置窗口，选择取消的时候会出现两个错误提示窗

2、选择自定义波特率时，修改完成后，无法再二次自定义波特率。

### 复现步骤 / To Reproduce

选择自定义波特率

### 预期的行为 / Expected behavior

1、一般情况下，选择取消按钮时不应该有任何错误提示。

2、当自定义波特率修改完成后，应该还需要可以再次自定义

### 系统 / System

Windows 10

### 软件版本 / Version

1.1.3.3

### 验证

- [X] 检查该问题是否已被提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.

LuaRobot, 2024-12-29:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction
- ❌未提供出错时需要的日志文件 does not provide logs when bug happening

mt6595, 2025-01-04:
将原来的SelectionChanged事件处理替换为DropDownOpened事件和DropDownClosed事件

private void BaudRateComboBox_DropDownClosed(object sender, EventArgs e)
{
    if (baudRateComboBox.SelectedItem != null)
    {
        if (baudRateComboBox.SelectedIndex == baudRateComboBox.Items.Count - 1)
        {
            int br = 0;
            Tuple<bool, string> ret = Tools.InputDialog.OpenDialog(TryFindResource("ShowBaudRate") as string ?? "?!",
                "115200", TryFindResource("OtherRate") as string ?? "?!");

            if (ret.Item1 && int.TryParse(ret.Item2, out br))
            {
                Tools.Global.setting.baudRate = br;
            }

            Task.Run(() =>
            {
                this.Dispatcher.Invoke(new Action(delegate {
                    var text = Tools.Global.setting.baudRate.ToString();
                    baudRateComboBox.Items[baudRateComboBox.Items.Count - 1] = text;
                    baudRateComboBox.Text = text;
                }));
            });
        }
        else
        {
            Tools.Global.setting.baudRate =
                int.Parse((baudRateComboBox.SelectedItem as ComboBoxItem).Content.ToString());
            baudRateComboBox.Items[baudRateComboBox.Items.Count - 1] = TryFindResource("OtherRate") as string ?? "?!";
        }
    }
}

private void BaudRateComboBox_DropDownOpened(object sender, EventArgs e)
{
    baudRateComboBox.Items[baudRateComboBox.Items.Count - 1] = TryFindResource("OtherRate") as string ?? "?!";
}

mt6595, 2025-01-04:
将原来的SelectionChanged事件处理替换为DropDownOpened事件和DropDownClosed事件
```c#
private void BaudRateComboBox_DropDownClosed(object sender, EventArgs e)
{
    if (baudRateComboBox.SelectedItem != null)
    {
        if (baudRateComboBox.SelectedIndex == baudRateComboBox.Items.Count - 1)
        {
            int br = 0;
            Tuple<bool, string> ret = Tools.InputDialog.OpenDialog(TryFindResource("ShowBaudRate") as string ?? "?!",
                "115200", TryFindResource("OtherRate") as string ?? "?!");

            if (ret.Item1 && int.TryParse(ret.Item2, out br))
            {
                Tools.Global.setting.baudRate = br;
            }

            Task.Run(() =>
            {
                this.Dispatcher.Invoke(new Action(delegate {
                    var text = Tools.Global.setting.baudRate.ToString();
                    baudRateComboBox.Items[baudRateComboBox.Items.Count - 1] = text;
                    baudRateComboBox.Text = text;
                }));
            });
        }
        else
        {
            Tools.Global.setting.baudRate =
                int.Parse((baudRateComboBox.SelectedItem as ComboBoxItem).Content.ToString());
            baudRateComboBox.Items[baudRateComboBox.Items.Count - 1] = TryFindResource("OtherRate") as string ?? "?!";
        }
    }
}

private void BaudRateComboBox_DropDownOpened(object sender, EventArgs e)
{
    baudRateComboBox.Items[baudRateComboBox.Items.Count - 1] = TryFindResource("OtherRate") as string ?? "?!";
}
```

closed 2025-01-21 by chenxuuu