chenxuuu/llcom #187: 在开启终端模式，显示框获取/失去焦点时会反复横跳

Iezuni, 2025-02-26:
### 描述一下这个bug / Describe the bug

点一下显示框显示ui就会有一次抖动

[image]

### 复现步骤 / To Reproduce

点一下显示框,再点一下其他的输入框

### 预期的行为 / Expected behavior

不再抖动

### 系统 / System

win10

### 软件版本 / Version

1.139

### 验证

- [x] 检查该问题是否已被提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [x] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.

LuaRobot, 2025-02-26:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction
- ❌未提供出错时需要的日志文件 does not provide logs when bug happening

chenxuuu, 2025-02-26:
设计如此

Iezuni, 2025-02-26:
抖着有点怪

chenxuuu, 2025-02-26:
影响不大，先合并了

感谢pr

closed 2025-02-26 by chenxuuu