chenxuuu/llcom #189: 串口列表刷不出来

sunniexx, 2025-02-27:
### 描述一下这个bug / Describe the bug

我同事用llcom时串口列表刷不出来，经排查，是卡在从Win32_PnPEntity获取串口的步骤，删掉这部分就好了。
是否可以考虑加一个重试次数，超过一定次数之后，以后就跳过这一部分？

### 复现步骤 / To Reproduce

打开llcom，串口列表一直是空的

### 预期的行为 / Expected behavior

正常应该会刷出串口列表

### 系统 / System

Win10

### 软件版本 / Version

1.1.3.9

### 验证

- [x] 检查该问题是否已被提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [x] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.

LuaRobot, 2025-02-27:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction
- ❌未提供出错时需要的日志文件 does not provide logs when bug happening

chenxuuu, 2025-02-27:
提供一下设备管理器该设备的截图吧