chenxuuu/llcom #190: 能否做一个分包,但是不显示时间戳的功能

Iezuni, 2025-03-01:
### 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.

现在的版本下,如果不开启分包功能,就无法触发接收脚本, 但是开启分包就会在每条数据前有时间戳,而且无法完整复制,

### 描述您想要的解决方案 / Describe the solution you'd like

分包,但是不显示时间戳的功能

### 描述您想要的详细使用步骤描述 / Describe the solution you'd like to use in what way

分包,但是不显示时间戳的功能

LuaRobot, 2025-03-01:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供功能需求的详细描述 does not provide detailed description of the feature request

Iezuni, 2025-03-02:
从代码上来看,不分包就是不会触发的
https://github.com/chenxuuu/llcom/blob/080a6684c707d8b31967249fc360d185b6b711fd/llcom/Pages/DataShowPage.xaml.cs#L79
https://github.com/chenxuuu/llcom/blob/080a6684c707d8b31967249fc360d185b6b711fd/llcom/Pages/DataShowPage.xaml.cs#L97

chenxuuu, 2025-03-02:
那可能当时设计的就是这样？
从直觉上来说不分包确实也没有处理的必要，用户自己判断分包截断太麻烦了

   

 https://github.com/chenxuuu/llcom/blob/080a6684c707d8b31967249fc360d185b6b711fd/llcom/Pages/DataShowPage.xaml.cs#L79
 https://github.com/chenxuuu/llcom/blob/080a6684c707d8b31967249fc360d185b6b711fd/llcom/Pages/DataShowPage.xaml.cs#L97

 
 https://github.com/chenxuuu/llcom/blob/080a6684c707d8b31967249fc360d185b6b711fd/llcom/Pages/DataShowPage.xaml.cs#L79
 https://github.com/chenxuuu/llcom/blob/080a6684c707d8b31967249fc360d185b6b711fd/llcom/Pages/DataShowPage.xaml.cs#L97

Iezuni, 2025-03-02:
所以要不要考虑一下我一开始提出的建议