chenxuuu/llcom #194: 分文件夹保存的llcom还是不能同事同时打开

sum411120, 2025-04-09:
### 描述一下这个bug / Describe the bug

分文件夹保存的llcom还是不能同事同时打开

### 复现步骤 / To Reproduce

新建文件夹保存llcom

### 预期的行为 / Expected behavior

应该可以同时使用

### 系统 / System

Win10

### 软件版本 / Version

1

### 验证

- [x] 检查该问题是否已被提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [x] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.

LuaRobot, 2025-04-09:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction
- ❌未提供出错时需要的日志文件 does not provide logs when bug happening

chenxuuu, 2025-04-09:
单独打开每个exe再关闭，每个都这样操作一遍

sum411120, 2025-04-09:
可以了👍🏻

来自 meizu 17

-------- 原始邮件 --------
发件人：chenxuuu ***@***.***>
时间：2025年4月9日 12:55
收件人：chenxuuu/llcom ***@***.***>
抄送：sum411120 ***@***.***>,Author ***@***.***>
主题：Re: [chenxuuu/llcom] 分文件夹保存的llcom还是不能同事同时打开 (Issue #194)

closed 2025-04-09 by chenxuuu